beyondscreen/node-rpi-ws281x-native #6: investigate problem with short typed-arrays

usefulthink, 2015-03-03:
quoting from #5:
> One thing I have noticed is that if you new up an Uint32Array with length 15 or under, it balks in the 
> C++ code claiming it's not a Uint32Array. This might be an internal node/io.js "optimisation" though.

achingbrain, 2015-03-03:
To reproduce:

```javascript
var strip = require('rpi-ws281x-native')

try {
  var numLeds = 20

  console.info('strip init')
  strip.init(numLeds)

  setInterval(function() {
    numLeds--
    console.info('using', numLeds)

    var data = new Uint32Array(numLeds)

    for(var i = 0; i < numLeds; i++) {
      var value = Math.round(Math.random() * 255) +
        (Math.round(Math.random() * 255) << 8) + 
        (Math.round(Math.random() * 255) << 16)

      data[i] = value
    }

    console.info('strip render')
    strip.render(data)
  }, 1000)
} catch(e) {
  console.error(e.message)
  console.error(e.code)
}
```

Console output:

```
# node index.js 
strip init
using 19
strip render
using 18
strip render
using 17
strip render
using 16
strip render
/home/pi/lightbot/index.js:24
    strip.render(data)
          ^
TypeError: render(): expected argument to be an Uint32Array.
    at TypeError (native)
    at null.<anonymous> (/home/pi/lightbot/index.js:24:11)
    at wrapper [as _onTimeout] (timers.js:243:14)
    at Timer.listOnTimeout (timers.js:88:15)
```

usefulthink, 2015-03-03:
doesn't happen in `node-0.10.33`. Which version are you running? 

A guess would be that V8 stores the data for very small typed-arrays inline (so it doesn't have any external array attached). Question would be how to check for this case.

achingbrain, 2015-03-03:
I'm running io.js 1.4.2.  I see the same behaviour on the pi with node 0.12.0 but not 0.10.36.

usefulthink, 2015-03-03:
OK. I will habe a look at v8, there probably is a better way to check for typed-arrays and get a pointer to the underlying array.

usefulthink, 2015-04-05:
with the changes from #15 this should be resolved now (v0.4.0 on npm)

closed 2015-04-05 by usefulthink