beyondscreen/node-rpi-ws281x-native #11: build instructions and verification for RPi2

usefulthink, 2015-03-31:
Apparently, there is an updated version of the rpi_ws281x-library which also supports the RPi2 here: https://github.com/richardghirst/rpi_ws281x

In order to support the RPi2 in this module we need to do the following:
 - [ ] test if that new version also works with the RPi1.
 - [ ] update the module to use the new version.

Unfortunately, I don't own a RPi2, so I need someone else to test this.

## instructions

> (all of this should be done on the raspberry-pi)

First, you'll need a clone of this repository.

    git clone --recursive https://github.com/raspberry-node/node-rpi-ws281x-native.git
    cd rpi-ws281x-native

And verify that the build works as it is:

    npm install -g node-gyp
    node-gyp rebuild

Switch the submodule to the forked library:

 * edit `.gitmodules` to contain the git-url of the forked repository, should read

   ```
   [submodule "src/rpi_ws281x"]
	    path = src/rpi_ws281x
	    url = git@github.com:richardghirst/rpi_ws281x.git
   ```

 * next, we need to tell git to update the submodule content

   ```
   git submodule sync
   git submodule update --remote --force
   ```

   at this point git should be telling you something like

   ```
   remote: Counting objects: 19, done.
   remote: Compressing objects: 100% (4/4), done.
   remote: Total 19 (delta 5), reused 5 (delta 5), pack-reused 10
   Unpacking objects: 100% (19/19), done.
   From github.com:richardghirst/rpi_ws281x
   de5a765..39afaac  master     -> origin/master
   Submodule path 'src/rpi_ws281x': checked out '39afaac5f2b8b307d7d7b5f2f790fbb6759bda5e'
   ```
   
   check for the commit-id in the last line, if that one matches, everything worked fine.

 * from what i saw, the forked library introduced some new C-files, so you need to edit the `binding.gyp`, which contains the recipes for building the library and the node-module. The interesting section is the `sources`-list in lines 16-20. Might be as easy as adding the new files (`board_info.c` and `mailbox.c`).
 * you can always check if it works by running `node-gyp rebuild`
 * if it works, there is one more thing left: in [lib/ws281x-native.js](https://github.com/raspberry-node/node-rpi-ws281x-native/blob/master/lib/ws281x-native.js#L3) there is a routine that checks the CPU-ID before the module is loaded. This should be extended to include the CPU-ID of the RPi2 (have a look at the output of `cat /proc/cpuinfo`, the line should start with "Hardware:").

Finally: Send a PR.
Thanks a lot for your help!

gavspav, 2015-04-01:
Once I installed nan via npm I built the original version.
I've edited .git modules but I'm now stuck on 'git submodule update --remote --force' Doesn't seem to work. The pi spits out usage rules for git submodule update. Remote and force don't seem to be amongst the options. Maybe I need to update?

usefulthink, 2015-04-01:
hm, yes. There were some changes to how git handles submodules in recent versions. 
You can try to update it manually, by changing into the submodule, update the remote and pull from the new master-branch. So, something like

```
cd src/rpi_ws281x
git remote set-url origin git@github.com:richardghirst/rpi_ws281x.git
git checkout master
git pull
```

gavspav, 2015-04-01:
Hmm - a load of stuff I don't really understand:

gyp info spawn args   '-Goutput_dir=.' ]
Traceback (most recent call last):
  File "/usr/local/lib/node_modules/node-gyp/gyp/gyp_main.py", line 18, in <module>
    sys.exit(gyp.script_main())
  File "/usr/local/lib/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 534, in script_main
    return main(sys.argv[1:])
  File "/usr/local/lib/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 527, in main
    return gyp_main(args)
  File "/usr/local/lib/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 503, in gyp_main
    options.circular_check)
  File "/usr/local/lib/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 129, in Load
    params['parallel'], params['root_targets'])
  File "/usr/local/lib/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 2692, in Load
    variables, includes, depth, check, True)
  File "/usr/local/lib/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 362, in LoadTargetBuildFile
    includes, True, check)
  File "/usr/local/lib/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 219, in LoadOneBuildFile
    None)
  File "binding.gyp", line 19
    './src/rpi_ws281x/board_info.c’,
                                     ^
SyntaxError: EOL while scanning string literal
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/node-gyp/lib/configure.js:343:16)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Linux 3.18.7-v7+
gyp ERR! command "node" "/usr/local/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_interface/node-rpi-ws281x-native
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok

usefulthink, 2015-04-01:
that sounds like something is wrong with the `binding.gyp`-file.

gavspav, 2015-04-01:
I thought that too but I couldn't see anything.Tried again and I used sublime text instead of textedit. Now I get this:

../src/rpi_ws281x/ws2811.c: In function ‘ws2811_init’:
../src/rpi_ws281x/ws2811.c:473:5: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
../src/rpi_ws281x/ws2811.c:478:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/rpi_ws281x/ws2811.c: In function ‘ws2811_render’:
../src/rpi_ws281x/ws2811.c:617:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  CC(target) Release/obj.target/libws2811/src/rpi_ws281x/pwm.o
  CC(target) Release/obj.target/libws2811/src/rpi_ws281x/dma.o
../src/rpi_ws281x/dma.c:46:1: warning: ‘static’ is not at beginning of declaration [-Wold-style-declaration]
make: *** No rule to make target 'Release/obj.target/libws2811/src/rpi_ws281x/malbox.o', needed by 'Release/obj.target/ws2811.a'.  Stop.
make: Leaving directory '/home/pi/node_interface/node-rpi-ws281x-native/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Linux 3.18.7-v7+
gyp ERR! command "node" "/usr/local/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_interface/node-rpi-ws281x-native
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok

usefulthink, 2015-04-01:
ah, alright. Then you were probably having errors with line-endings (`\n` vs. `\r\n`),
the warnings are kinda ok, the main error-message is a bit hard to spot:

> No rule to make target 'Release/obj.target/libws2811/src/rpi_ws281x/malbox.o'

which pretty much looks like a typo (otherwise i'd have no idea where `malbox.o` should be coming from.

gavspav, 2015-04-01:
God I'm such a twat - sorry! and thanks for sticking with me. 
I'll just edit the .js function and report back. 
What does 'Finally: Send a PR.' mean?

usefulthink, 2015-04-01:
a pull-request, so i can integrate your work into this repository (see here https://guides.github.com/introduction/flow/).

gavspav, 2015-04-01:
Well index.js runs ok but rainbow.js in the examples throws:

Failed to create mailbox device /tmp/mailbox-9651: Operation not permitted
Failed to create mailbox device /tmp/mailbox-9651: Operation not permitted
Failed to create mailbox device /tmp/mailbox-9651: Operation not permitted
/home/pi/node_interface/node-rpi-ws281x-native/lib/ws281x-native.js:94
    ws281xNative.init(numLeds, options);
                 ^
Error: init(): initialization failed. sorry – no idea why.

So I tried it as root and got this:

Press <ctrl>+C to exit.
/home/pi/node_interface/node-rpi-ws281x-native/lib/ws281x-native.js:129
    ws281xNative.render(data);
                 ^
TypeError: render(): expected argument to be an Uint32Array.
    at TypeError (native)
    at EventEmitter.ws281x.render (/home/pi/node_interface/node-rpi-ws281x-native/lib/ws281x-native.js:129:18)
    at null.<anonymous> (/home/pi/node_interface/node-rpi-ws281x-native/examples/rainbow.js:23:10)
    at wrapper [as _onTimeout] (timers.js:265:14)
    at Timer.listOnTimeout (timers.js:110:15)

I haven't actually got the lights connected yet - doesn't look like that is the issue but better check?

gavspav, 2015-04-01:
Ah ok I need arguments don't I?

usefulthink, 2015-04-01:
ah, yes. You definitely need to run the script as root, otherwise you don't have access to the memory-mapped I/O. 

There currently is a bug with newer node/iojs-versions and I'm working on a fix (see #6). The single argument for the rainbow.js is the number of LEDs, which has to be at least 16 (if you have fewer LEDs that doesn't matter). So calling `node examples/rainbow.js 20` should at least not throw any errors.

gavspav, 2015-04-01:
Yeah - give us half an hour an I'll work out how to do a Pull Request!
Then I'll try and connect the lights!

gavspav, 2015-04-01:
I think I've done the Pull Request!
No idea if the forked library works with the original Pi though.
The .js function checks for BCM2708 or 9 so if it doesn't work with the
original it will be a bit misleading.
And in any case there is probably a better way of doing the 8 or 9 thing.
Thanks for walking me through it all - gonna try and connect the lights
next.
Gavin

usefulthink, 2015-04-01:
Ok, I'll try that with the old Pi later today and see if it works.

The Pull-Request looks good. Can you report back when you get the lights working?

gavspav, 2015-04-01:
Yup Lights working here.

usefulthink, 2015-04-02:
ok, i think i found a working solution to have full interoperability with both RPi 1 and 2 models. Can be found in #13. I will do some tests later today and update the npm-package. 

@gavspav it would be great if you could test if everything works on the RPi2 when the new Version is on npm.

gavspav, 2015-04-02:
Yeah for sure. Give me shout when you've done it.

usefulthink, 2015-04-02:
ok, i just published the new 0.3-version, which works on both the old and (in theory) the new raspberry. @gavspav Can you please run the test to see if it works for you?

So if you run `npm install rpi-ws281x-native` in an empty directory, the example should work using `node node_modules/rpi-ws281x-native/examples/rainbow.js 20`.

gavspav, 2015-04-02:
Hmm the Pi is at my Studio and I'm going away for a couple of days so won't
be able to test until Saturday or Sunday.
Sorry
G

stefancarlton, 2015-04-04:
Hi,

I'm running a Pi2 and checked this code out - it works albeit I had to make the quick-fix found in PR #14.

Thanks for your work @gavspav and @usefulthink :)

usefulthink, 2015-04-05:
thank you very much. I guess this can be closed now :)

closed 2015-04-05 by usefulthink