beyondscreen/node-rpi-ws281x-native #16: Conflict with Pure Data

gavspav, 2015-04-15:
I'm trying to run the library in conjunction with Pd-extended (https://puredata.info) on the RPI 2.

When I open Pure Data on the Pi the rainbow.js patch stops working properly.

If I close it and start it again I get some leds flickering, seemingly randomly.

Closing Pure Data does not solve this. I have to restart the Pi before the rainbow patch will run properly.

I have tried assigning each process to a separate core, using taskset, but it makes no difference.

Any way around this?

usefulthink, 2015-04-15:
I haven't heard of puredata yet, but from a quick glance at it's sourcecode it looks like a known Problem: somehow (I haven't dug into the exact reasons yet) the RPi internal "soundcard" uses the same GPIO / DMA / PWM functions that are needed to run the LED-drivers. 
So from what I know you can not use both at the same time. 

I heard of some possible solutions (but, as i don't do any audio haven't tested any of them): 

 * using some simple USB Soundcard instead of the internal audio-device might help
 * disabling the raspberry internal audio if you don't need it (you might need to disable/blacklist the corresponding `snd_*` kernel-modules to do this)
 * i somehow recall that doing audio via hdmi-out also doesn't have this problem, but not sure about this

(i currently don't have the time to dig into this, but it looks like [hyperion](https://github.com/tvdzwan/hyperion) is successfully running an ambilight-implementation based on ws28xx-LEDs in parallel to video/sound output)

gavspav, 2015-04-15:
Thanks - yes that looks like the culprit as disabling the internal
soundcard works.

I'm looking for a workaround - I'll post it here if and when I find one.

gavspav, 2015-04-15:
I have an external soundcard so I tried commenting out the sound module in /etc/modules but this seemed to disable all alsa devices.
So I edited /etc/modprobe.d/alsa-base.conf to allow my external soundcard to be the first device.
This seems to work for me - sound and lights!

usefulthink, 2015-04-15:
we should probably update the README to contain this as a known issue.

usefulthink, 2015-04-15:
otherwise, awesome - great to know that this actually works. Would you mind writing a few lines for the README?

gavspav, 2015-04-15:
Yeah for sure. I'll do it tonight.

dlopuch, 2015-12-03:
Hi!  Came across this library from http://hackaday.com/2015/12/02/the-led-roundsystem

It seems the sound conflict comes from using rpi_ws281x, which uses the rpi's PWM to generate signal timing for the LEDs.  An alternative could be to use SPI.  I've [used SPI in node to drive some LPD8806's](https://github.com/dlopuch/glow-ops/blob/master/lib/controllers/lightstrip.js) in a weekend hack once.

Advantages are SPI is just a file stream at `/dev/spidev0.0, so no special C extensions required.  Since it's a different piece of hardware, I don't think it conflicts with the sound hardware like the sole PWM does.  The pi has an SPI driver, but it's not always enabled by default (though distros like adafruit's rasbian do enable it by default).

Not sure about the timing differences between WS281x's vs. LPD8806's, but SPI could provide another way of doing it.  Hope that helps!

usefulthink, 2015-12-03:
Hi @dlopuch, thank you very much for your reply and this great Idea! 
I like how simple it actually is to drive these LPD8806-LEDs. 
If you are interested, we could consider to add support for these lightstrips into this library.

However, that SPI works for driving those LEDs is mainly because they are using a serial protocol with seperate clock and data-lines. This is not the case for the ws281x-LEDs, which use a single data-line (implicit clock) and a pulsewidth-encoding for binary-values. They also quite timing-sensitive with tolerances of around 0.3µs.

I still need to do a bit of reading about SPI usage, but from my current understanding it will at best be difficult to get this going. One way I can imagine this would work, would be to have SPI clocked at 2.4MHz and just use the MOSI data-line, similar to how we are currently using the PWM-module. But I need to have a look with my oscilloscope to see if this could work.