beyondscreen/node-rpi-ws281x-native #18: Pin 13 on Rpi b+ and a+ broken

menkveldj, 2015-05-07:
Hi

It looks like the library is having trouble initializing the new PWM Pin 13 on the a+ and b+.

Im working on a project that would utilizes both 18 and 13. Perhaps this is an easy fix?

Thanks

usefulthink, 2015-05-09:
currently this module only supports the PWM0 on GPIO18, but the underlying library already supports both PWM-controllers (see here: https://github.com/jgarff/rpi_ws281x/issues/30). I will probably have some time to experiment with this in the next weeks, but currently it doesn't look like there is a dead easy fix for this. We would at least need to make it possible to use multiple channels from the node-module.

menkveldj, 2015-05-09:
> jgarff/rpi_ws281x#30 <https://github.com/jgarff/rpi_ws281x/issues/30>). I

Cool thanks for the update. Look forward to seeing what you come up with.

usefulthink, 2015-06-01:
Quick update: I have just ordered a rpi2, so i will be able to test how everything works with the second PWM and the corresponding GPIO.

mweinelt, 2016-11-01:
Any update on this?

usefulthink, 2016-11-01:
Sadly no. I didn't find the time yet to refactor the library to support multiple pwm-outputs. If you would like to help, I already made a plan about what needs to be done and could help you getting this to work.

b33son, 2016-11-07:
Over at Falcon, they've produced a Pi Hat & a Pi Cap with two outputs. I can verify this library works great with the Pi Hat on output 1 (GPIO 18). Even if only the one output works. It's still freakin awesome. I'm sure peeps would find the second output very useful.

damianprzygodzki, 2016-12-11:
Any updates? Would be great to init two instances of this at once.

usefulthink, 2017-12-01:
updates here: #67