beyondscreen/node-rpi-ws281x-native #19: Root required

akbsteam, 2015-05-10:
Hi,

Seems to require running with sudo e.g.:

    sudo node node_modules/rpi-ws281x-native/examples/rainbow.js 190

How can I run it as my normal user - gpio-admin or something similar (didn't seem to work for me)

usefulthink, 2015-05-11:
Actually, i haven't yet bothered to find out and I currently have no raspberry here to test.

As far as I know, the only thing in the driver that requires elevated privileges is access to the physical memory via `/dev/mem`, so you could try to allow access to that device-file for another user.

usefulthink, 2015-05-11:
I now did have a look at it and unfortunately it will not really be possible to run this module without root-privileges, there is [an answer on stackoverflow](http://stackoverflow.com/a/22370947/123219) that goes into some details as to why it is a bad idea to allow anyone than root access to `/dev/mem` (and even if you change permissions on that device-file anyway, it still won't work - physical memory is somewhat special i guess).

Access to that file is absolutely mandatory, as the driver-part needs access to some memory-mapped CPU-Functions like the PWM-modules and DMA Control-Registers that cannot be addressed from the usual virtual memory a userspace-process gets to see.

If there are parts of your program you don't want to have running as root, the best thing you could do is probably something like splitting the LED-Driver out into a seperate-process and send data to it using streams. You could for instance use the open pixel-control protocol to do this: https://github.com/raspberry-node/node-openpixelcontrol

closed 2015-05-11 by usefulthink

usefulthink, 2015-05-12:
Another update: It seems that you can disable the mmap-protection if you compile the kernel yourself. 

http://stackoverflow.com/questions/11891979/accessing-mmaped-dev-mem
http://lwn.net/Articles/267427/

SteveEdson, 2015-10-01:
I believe the latest version of raspbian allows sudo-less access to the GPIO pins, would this solve the issue?

usefulthink, 2015-10-02:
unfortunately it's not just access to the GPIO-Pins, but access to the raw physical memory that is the problem here. I haven't yet found a way to configure the DMA and PWM modules of the Pi's CPU without going through the `/dev/mem` device, which is what we need root-access for.