beyondscreen/node-rpi-ws281x-native #21: Autorun script

gavspav, 2015-05-14:
Not an issue with the library but I'm struggling to run my script automatically on boot.

Perhaps it would be useful for people if anyone could help me nail it!

I currently have sudo node /myscript.js & in a script which gets executed via rc.local.

Top shows that node is running, but the lights don't work. I'm guessing its something to do with permissions.

Zefiro, 2015-05-14:
It also might have something to do with $path being initialized differently when run from rc.local
I'd suggest using full path names.

Myself, I'm using this setup:

added to end of **/etc/rc.local**
> \# start appliance
> su -c /home/zefiro/bin/start_screen.sh zefiro

**/home/zefiro/bin/start_screen.sh** (which is chmod u+x)
> \#!/bin/bash
> cd /home/zefiro/prog/wodo-raspi-light
> screen -AdmS slave -t run
> screen -S slave -X screen -t edit1
> screen -S slave -X screen -t edit2
> screen -S slave -p run -X stuff $'sudo node ledstrip.js\n'
> screen -S slave -p edit1 -X stuff $'nano ledstrip.js \n'

I also set it such that when I log in, I'm directly put into this screen. Much easier for debugging (seing the console output).

usefulthink, 2015-05-27:
You should have a look at pm2, which can do a lot of things (like monitoring your script, restart if it crashes, etc) and can also install the necessary files to run on system startup.

https://github.com/Unitech/PM2
https://www.digitalocean.com/community/tutorials/how-to-use-pm2-to-setup-a-node-js-production-environment-on-an-ubuntu-vps

closed 2015-06-10 by usefulthink