beyondscreen/node-rpi-ws281x-native #22: Best way to change to rgb output?

gavspav, 2015-05-24:
This is from an email but it would be better on here.

Ages ago I dug deep into the library and swapped the r and the g of the rgb around so that it would render correctly on my grb led's!

Now I'm using a different installation and can't remember how the hell I did it! I think that proves I am degenerating. Any idea where to look? You might have even told me in the first place!

I'm sure I'll find it - when I do this would probably be useful for the read me? 

Gavin

As the underlying driver still lacks support for GRB-LEDs (https://github.com/jgarff/rpi_ws281x/issues/17)  I think it would be best to simply manipulate the Uint32Array in Javascript before it get passed to the native driver.
For this, the best option would be to use the "beforeRender"-event fired here https://github.com/raspberry-node/node-rpi-ws281x-native/blob/master/lib/ws281x-native.js#L146

Something like this:

var ws281x = require('rpi-ws281x-native');

ws281x.on('beforeRender', function(arr) {
  // do whatever you want with the pixel-data in arr
});

Hope it helps.
Martin

gavspav, 2015-05-24:
Hmm ok. Is that not quite a costly operation though going through the whole array every frame?

Do you think it's worth the effort trying it at a lower level?

usefulthink, 2015-05-25:
Regarding performance: for typed arrays and a constant access-pattern (read int, modify, write back, no type coercions etc.) v8 is able to optimize this a lot. And it's still just a loop over each LED (which would very rarely exceed 500 or so), so i guess that should be manageable performance-wise.

I would like to see changes like these in the underlying library, where all that bit-shifting into the output-buffers has to happen anyway. The change there would probably be minimal (add a new setting `colorOrder` to the `ws2811_channel_t`-struct and change the color-value generation in https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.c#L697-L702 accordingly).
 
That would probably be worth the effort, but we probably would have to maintain our own fork of the library as the original author seems to have abandoned the project.

usefulthink, 2015-05-27:
I just removed the original version of the library (see #23) as i noticed that the newer version by richardghirst also supports the first generation raspberries.

So implementing something like different LED-types should be based on that version of the lirbary.

usefulthink, 2017-12-03:
for reference, there is now a first working version of the upcoming 1.0-version that supports all different strip-types. See #67 for all the details.

closed 2017-12-03 by usefulthink