beyondscreen/node-rpi-ws281x-native #27: Cannot build on RPi B+

mrminecart, 2015-06-18:
When executing the build command with the verbose option

```
pi@raspberrypi /var/node/LED-Stips/hello/node-rpi-ws281x-native $ sudo node-gyp rebuild -v
gyp info it worked if it ends with ok
gyp verb cli [ 'node', '/usr/local/bin/node-gyp', 'rebuild', '-v' ]
gyp info using node-gyp@2.0.1
gyp info using node@0.10.32 | linux | arm
gyp verb command rebuild []
gyp verb command clean []
gyp verb clean removing "build" directory
gyp verb command configure []
gyp verb check python checking for Python executable "python2" in the PATH
gyp verb `which` succeeded python2 /usr/bin/python2
gyp verb check python version `python2 -c "import platform; print(platform.python_version());"` returned: "2.7.3\n"
gyp verb get node dir no --target version specified, falling back to host node version: v0.10.32
gyp verb command install [ 'v0.10.32' ]
gyp verb install input version string "v0.10.32"
gyp verb install installing version: 0.10.32
gyp verb install --ensure was passed, so won't reinstall if already installed
gyp verb install version is already installed, need to check "installVersion"
gyp verb got "installVersion" 9
gyp verb needs "installVersion" 9
gyp verb install version is good
gyp verb get node dir target node version installed: 0.10.32
gyp verb build dir attempting to create "build" dir: /var/node/LED-Stips/hello/node-rpi-ws281x-native/build
gyp verb build dir "build" dir needed to be created? /var/node/LED-Stips/hello/node-rpi-ws281x-native/build
gyp verb build/config.gypi creating config file
gyp verb build/config.gypi writing out config file: /var/node/LED-Stips/hello/node-rpi-ws281x-native/build/config.gypi
gyp verb config.gypi checking for gypi file: /var/node/LED-Stips/hello/node-rpi-ws281x-native/config.gypi
gyp verb common.gypi checking for gypi file: /var/node/LED-Stips/hello/node-rpi-ws281x-native/common.gypi
gyp verb gyp gyp format was not specified; forcing "make"
gyp info spawn python2
gyp info spawn args [ '/usr/local/lib/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/var/node/LED-Stips/hello/node-rpi-ws281x-native/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.node-gyp/0.10.32/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.node-gyp/0.10.32',
gyp info spawn args   '-Dmodule_root_dir=/var/node/LED-Stips/hello/node-rpi-ws281x-native',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp verb command build []
gyp verb build type Release
gyp verb architecture arm
gyp verb node dev dir /root/.node-gyp/0.10.32
gyp verb `which` succeeded for `make` /usr/bin/make
gyp info spawn make
gyp info spawn args [ 'V=1', 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/var/node/LED-Stips/hello/node-rpi-ws281x-native/build'
make: *** No rule to make target 'Release/obj.target/rpi_libws2811/src/rpi_ws281x/ws2811.o', needed by 'Release/obj.target/rpi_libws2811.a'.  Stop.
make: Leaving directory '/var/node/LED-Stips/hello/node-rpi-ws281x-native/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:810:12)
gyp ERR! System Linux 3.12.29+
gyp ERR! command "node" "/usr/local/bin/node-gyp" "rebuild" "-v"
gyp ERR! cwd /var/node/LED-Stips/hello/node-rpi-ws281x-native
gyp ERR! node -v v0.10.32
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
```

System:
```
$ cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
NAME="Raspbian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

usefulthink, 2015-06-19:
First of all, I dropped support for node <=0.12 a while ago. I'd recommend upgrading to node-0.12, but otherwise you can stick to the v0.3.0-version of this module which is the last that had support for the node-0.10-branch.

I tried to reproduce your problem, and I get the very same error-message when the git submodules are not properly intialized.  
Can you please check if the submodules are initialized in your working-directory (is there a file `src/rpi_ws281x/ws2811.c`?) If there isn't that's likely you problem. Assuming you have a fresh checkout of the repository there, simply run `git submodule update --init` to fetch the contents of the submodule.

mrminecart, 2015-06-21:
Ah okay,

Managed to build with node 0.12.4

closed 2015-06-22 by usefulthink