beyondscreen/node-rpi-ws281x-native #29: TODO for v1.0

usefulthink, 2015-06-25:
## node-module and native parts
 * [ ] support for multiple PWM channels (issue: #18)
 * [ ] support for different binary-formats (i.e. `CanvasPixelData`, `Uint8Array` and `Uint32Array`, `Buffer`/`ArrayBuffer`)
 * [ ] a tiny bit more abstraction: the public api should be fully implemented in Javascript, so the underlying native-implementation is able to move without breaking BC
 * [ ] generic implementation for per pixel and full data-array operations (color-correction, channel-splitting, whatever..)
 * [ ] switch to `node-pre-gyp` instead of `node-gyp`

## rpi_ws281x
   * [ ] start and maintain a fork based on richardghirst/rpi_ws281x
   * [ ] merge changes fixing `unmapmem`-calls (euang/rpi_ws281x@73997a5e33f0afdb8cc1f07f09a10a60e8866a9b)
   * [ ] merge optimizations for inverting level-shifters (archi/rpi_ws281x@e6ec6170d8ea3f5c2e4e633f4c6c8a43c0254024)
   * [ ] implement BGR/GRB color-formats (issues: #22, jgarff/rpi_ws281x#17)

usefulthink, 2016-02-04:
### changes required to support multiple channels

As the client-side code does not directly call the native implementations, it should be possible to change the implementation without breaking compatibility on the client-side.

 - [`<rpi-ws281x.cc>.init()`][1]
   - accept parameters for both channels, ideally two objects where the second one is optional, moving the ledCount into the object. 
   - initialize both channels before calling `ws281x_init()`
 - [`<rpi-ws281x.cc.render()`][2]
   - should accept two byte-arrays if both channels are used (this is preferable over accepting a channel-number b/c the call will always synchronously write out data for both channels

 - implementations of the javascript-bindings in [`ws281x-native.js`][3] should maintain their behavior but need to be update to the changes above.
 - there should be an updated version of the interface supporting multiple channels, this would be open for discussion how this could look.

[1]: https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/src/rpi-ws281x.cc#L61
[2]: https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/src/rpi-ws281x.cc#L32
[3]: https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/lib/ws281x-native.js#L121

ben0bi, 2016-10-20:
I could not install it with nodejs v6 and I could not install nodejs v4.
For me, it works with nodejs v0.12.

Would like to mark this as todo, compatibility for nodejs >= v4
(Or is it my system? I use the RetroPie-Distro and LXDE as window manager. Multipurpose is the key.)

Regards,
ben0bi

usefulthink, 2016-10-20:
@ben0bi haven't touched this project in a while so it can very wel be that something got a bit outdated. Can you provide some more details about that failed installation attempt? Are there compiler-errors or something like that?

usefulthink, 2016-10-20:
I moved this into a seperate issue.

usefulthink, 2017-12-01:
see #67

closed 2017-12-01 by usefulthink