beyondscreen/node-rpi-ws281x-native #30: Gamma correction done twice if source map is set

kunaakos, 2015-07-08:
I think [this line](https://github.com/raspberry-node/node-rpi-ws281x-native/blob/master/lib/ws281x-native.js#L100) should be:
```JavaScript
data[i] = _tmpData[indexMapping[i]];
```
Gamma correction is done in render()

usefulthink, 2015-07-09:
duplicate of #28

usefulthink, 2015-07-09:
the problem is, the gamma-correction is applied to the data-array that is most likely reused for rendering the next frame. Thus, if you don't write the whole data-array on every frame the initially set colors will fade to black over time (because the correction is applied over and over again).

tsuckow, 2015-08-09:
I don't think render should be doing any of this stuff or at the very least there should be a "renderRaw" method that doesn't do any of these transformations.

If you do wish to do this, you should have your own buffer that is created at init time and used as the scratch space.

usefulthink, 2015-08-10:
I agree. Wasn't the best Idea to have this in the core. Will be removed in the next version.

usefulthink, 2015-11-27:
With the rewrite of the native part for version 0.6 this is now working properly, as the data passed by the application is not modified anymore.

closed 2015-11-27 by usefulthink