beyondscreen/node-rpi-ws281x-native #35: Error when trying to run rainbow.js

gavspav, 2015-09-22:
Just trying to get some lights working on an image from https://ccrma.stanford.edu/~eberdahl/Satellite/ .It is basically raspbian as I understand it.

I've installed node.js - when I run the rainbow patch I get the following error (below). 

No idea if this is an issue with the library - probably not - but any idea whats going on?

ccrma@satellite ~/node_modules/rpi-ws281x-native/examples $ sudo node rainbow.js 40
Can't open device file /tmp/mailbox-3693: Permission denied
Can't open device file /tmp/mailbox-3693: Permission denied
Can't open device file /tmp/mailbox-3693: Permission denied
/home/ccrma/node_modules/rpi-ws281x-native/lib/ws281x-native.js:119
    bindings.init(numLeds, options);
             ^
Error: init(): initialization failed. sorry – no idea why.

usefulthink, 2015-09-22:
hmmm. That looks like you hit this issue: https://github.com/raspberry-node/rpi_ws281x/issues/2

Do you see a device-file named `/dev/vcio` on the raspberry?
Also, you could check if the kernel-version (get it with `uname -r`) is above 4.1.6. In that case this is exactly the issue and we need to patch the rpi_ws281x-module to make it work.

Unfortunately I won't find time for this until next week. But if you like to give it a try, i could make you a org-member and try to guide you.

gavspav, 2015-09-22:
> hmmm. That looks like you hit this issue: raspberry-node/rpi_ws281x#2

Hmm the kernel is reported as 3.18.11-v7+

I do have /dev/vcio yes.

I'm up for trying to fix it if you (and search engines) can guide me, yes!

gavspav, 2015-09-22:
Opening /dev/vcio in nano, it looks empty.

Also I installed the module from npm - is that the latest version? I noticed only the rainbow.js  script is in the example folder and not the new script you added.

Just wondering if there was anything important missing?

usefulthink, 2015-09-22:
the last published version on npm is quite old as i saw yesterday... you should definitely install it via git. 

I just updated the master-branch once again to reflect the latest changes to the module (this includes our own fork of rpi_ws281x-module so we can just integrate the fix proposed in pimoroni/unicorn-hat@5c03f81). 

Check it out using `git clone --recursive git@github.com:beyondscreen/node-rpi-ws281x-native.git`, so you also have the submodules initialized. (If you already did so, it's probably best to do it again because git submodules are really weird at times).

Then you need to have the `node-gyp`-package installed (`npm install -g node-gyp`), and you can trigger the compilation of the native part of the module with `node-gyp rebuild`. I would at first try to bake the patch from the unicorn-hat-library (see above) into the rpi_ws281x-module (in `src/rpi_ws281x/`). `node-gyp rebuild`, and see if that helps.

gavspav, 2015-09-22:
I tried to clone it earlier but it asked me for a isa_rsa.pub key or something.
Hopefully I just need to generate an ssh key.
I'll try it in a bit

usefulthink, 2015-09-22:
Yes, exactly. You need to create an ssh-key and store the public-part (usually id_rsa.pub) in your guthub-account. Then you should be able to clone the repository using the git@github.com URL.

gavspav, 2015-09-23:
Think I'm doing something wrong! I've got to the node-gyp rebuild stage but I'm getting the errors below.
The rpi-ws281x folder is in a folder called node_modules - does this matter? And I haven't tried to patch in the unicorn workaround yet!

module.js:338
    throw err;
          ^
Error: Cannot find module 'nan'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at [eval]:1:1
    at Object.exports.runInThisContext (vm.js:74:17)
    at Object.<anonymous> ([eval]-wrapper:6:22)
    at Module._compile (module.js:460:26)
    at evalScript (node.js:431:25)
    at startup (node.js:90:7)
gyp: Call to 'node -e "require('nan')"' returned exit status 1. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/node-gyp/lib/configure.js:355:16)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.18.11-v7+
gyp ERR! command "node" "/usr/local/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/ccrma/node_modules/node-rpi-ws281x-native
gyp ERR! node -v v0.12.6
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok

gavspav, 2015-09-23:
I've patched the mailbox.c file now but can't work out how to get the node-gyp rebuild to work!

Ah ok I've installed the nan package via npm and now it's working.

Running rainbow.js seems to be working - I've got no lights here - I'll have to try it tomorrow.

I've got a feeling a might have blown the output pin on the Pi when hotplugging the other day (it shutdown instantly) so not sure I'm going to be able to tell!

If it works you'll have to let me know how to push the mailbox.c file to the repository! Thanks for your help.

gavspav, 2015-09-24:
I can confirm it works. I've made a pull request with the new version of mailbox c. Will it still work with older kernels though? Maybe not. So it might need a bit of either/or code?

Thanks for your help!

usefulthink, 2015-09-25:
Thank you so much for this <3

I have both a raspberry-pi 1 and 2 with older kernel-versions here and will test the changes soon. But from what I saw it will fall back to the older mailbox-device mode that works on older versions. Expect a new release to npm next week.

usefulthink, 2015-10-03:
Soo, I finally managed to publish a new version containing the fix you made. You should now be able to use the version v0.5.0 from npm.

So, thanks again for helping me with this!

closed 2015-10-03 by usefulthink