beyondscreen/node-rpi-ws281x-native #39: Example uses index mapping.

Guuz, 2015-11-24:
Hello,

One of the examples isn't working for me because it's using a mapping:
https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/examples/iterate.js#L8

Especially for this example it's nice to have no mapping. That way you can see how your strips are connected (if you have multiple strips). And mappings are always specific for every user/implementation so maybe it's a good idea to leave them out for all examples.

What do you think?

I can make a PR if you want, though it's only 1 line ;-)

usefulthink, 2015-11-24:
Oh, yes. That was the intention behind this example, no idea how that change slipped in there. PR would be great, thanks :)

closed 2015-11-24 by usefulthink

joedeveloper, 2018-12-13:
For me it would be valuable to have an example of index mapping.
There are a number of possible solutions I could choose between in controlling my leds - I really like the 'idea' of index mapping, but without an example I would have to commit time I can't get back to understand, leverage and validate this functionality.

This becomes harder to justify as the last package update timestamp slips further into the past.

Love yall for making this in the first place, just saying that the work could benefit more people if it was easier to evaluate.

usefulthink, 2018-12-14:
Hey @joedeveloper, the problem is that I couldn't see a way that solves the index-mapping in a way that is really useful for more than just a very narrow set of use-cases (and pretty much useless for all others). This is why it was dropped entirely from the newer version of the library (see https://github.com/beyondscreen/node-rpi-ws281x-native/tree/v1.x).

I think it might be worthwhile to approach that problem seperately in another module or specific to the actual use-case. I have a few Ideas about how this could be done, but right now I am lacking the time to actually do it.

The reason why the update-time slips into the past is that a) it just works as it is and b) I don't find the time and resources to validate the new version (see above) with all the hardware it should work with.

I'm not trying to convince you to use this library over any other solution out there, but if you want to use node/javascript for your project, this might be one of the better and cheaper options. 

If you need any help or advice, please feel free to reach out! I can't promise much, but I will always at least try to help.

PR #40 (2015-11-24): Remove index mapping in example.
As agreed in #39 index mapping is not wanted in the examples.

This closes #39.
commit 02cdf264 (2015-11-24): Remove index mapping in example.

As agreed in #39 index mapping is not wanted in the examples.

This closes #39.