beyondscreen/node-rpi-ws281x-native #42: Throw error or return mock.

Guuz, 2015-11-24:
Hello,

I'm using this module in my app and stumbled upon the following.
During development I don't want to run as root (why should i?) and I cant really use this module anyway because it only works on a Pi.
The code [here](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/lib/ws281x-native.js#L4) logs to `strerr` and exits the process. This is not very friendly for a consumer.

I would propose we throw an error that the consumer can detect and ignore during development.
Maybe the [mock implementation](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/lib/ws281x-native.js#L50) could be made available to use too.

Curious to hear what your opinion is on this! I can work on a PR if we come to a solution.

usefulthink, 2015-11-25:
I like that Idea. I never ran into such a situation myself, but it makes complete sense to be a bit more friendly about that :) I would suggest to move the root-verification into the `getNativeBindings`-function, similar to what is happening with the raspberry-detection.

Guuz, 2015-11-26:
`getNativeBindings()` makes sense yes! I think it's very bad for a 3th party lib to call process.exit().

usefulthink, 2015-11-26:
You are right. Would be better to just throw an exception or handle it gracefully (like it is currently done for non-pi architectures). The consuming library will then either have to catch and handle it or will die anyway with an uncaught exception. Would be great of you could create a PR for this if you find the time.

Guuz, 2015-11-26:
I'm on it!

Guuz, 2015-11-26:
@usefulthink I changed the implementation a bit compared to what we talked about here. But i think it's better this way. I hope you agree! If not I'm happy to change it.

PR #43 (2015-11-27): Always return stubs instead of calling process.exit().
We are not very friendly to consumers by calling process.exit().
Instead we now log a warning and return the stubs for convenience.

This closes #42.

I know I proposed throwing an error. Though when I was halfway that change I thought it made more sense to always return the stubs. Most of the time people aren't developing on a Pi so they need stubs anyway. I think this is the most consumer friendly solution.

If we do opt for throwing an error I think we should at least expose the sub methods for consumers to use.

To give you an idea, th [... cut]
closed 2015-11-27 by usefulthink