beyondscreen/node-rpi-ws281x-native #45: supprot for Openwrt?

evanlitw, 2016-01-23:
Is it supprot for Openwrt?
Thanks

usefulthink, 2016-01-23:
As @jgarff already pointed out (jgarff/rpi_ws281x#57), it won't work on anything but the raspberry-pi. Apart from that, I see no reason why it shouldn't also work with openwrt, although I'm not sure if you have a build toolchain running in openWRT. This is required as this module will compile the libraries on install.

Have you tried running `npm install rpi-ws281x-native`? What is the output?

evanlitw, 2016-01-23:
Yes. but have more error 

> rpi-ws281x-native@0.8.1 install /usr/rpi_ws281x/node_modules/rpi-ws281x-native
> node-gyp rebuild

gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: unexpected eof
gyp ERR! stack     at decorate (/usr/lib/node_modules/node-gyp/node_modules/fstream/lib/abstract.js:67:36)
gyp ERR! stack     at Extract.Abstract.error (/usr/lib/node_modules/node-gyp/node_modules/fstream/lib/abstract.js:61:12)
gyp ERR! stack     at Extract._streamEnd (/usr/lib/node_modules/node-gyp/node_modules/tar/lib/extract.js:91:35)
gyp ERR! stack     at BlockStream.<anonymous> (/usr/lib/node_modules/node-gyp/node_modules/tar/lib/parse.js:51:8)
gyp ERR! stack     at BlockStream.emit (events.js:104:17)
gyp ERR! stack     at BlockStream._emitChunk (/usr/lib/node_modules/node-gyp/node_modules/tar/node_modules/block-stream/block-stream.js:203:10)
gyp ERR! stack     at BlockStream.resume (/usr/lib/node_modules/node-gyp/node_modules/tar/node_modules/block-stream/block-stream.js:58:15)
gyp ERR! stack     at Extract.Reader.resume (/usr/lib/node_modules/node-gyp/node_modules/fstream/lib/reader.js:249:34)
gyp ERR! stack     at Entry.<anonymous> (/usr/lib/node_modules/node-gyp/node_modules/tar/lib/parse.js:261:8)
gyp ERR! stack     at Entry.emit (events.js:104:17)
gyp ERR! System Linux 3.18.21
gyp ERR! command "node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /usr/rpi_ws281x/node_modules/rpi-ws281x-native
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v3.2.1
gyp ERR! not ok
npm ERR! Linux 3.18.21
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "rpi-ws281x-native"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE

npm ERR! rpi-ws281x-native@0.8.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the rpi-ws281x-native@0.8.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the rpi-ws281x-native package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls rpi-ws281x-native
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/rpi_ws281x/npm-debug.log

usefulthink, 2016-01-24:
well, that just says that the compilation of the native module fails for some reason. 
From that output I can't see what exactly is going wrong, but it could be that openWRT doesn't come with the required build-environment to compile the module. Can you maybe attach the `npm-debug.log` (simply put it into a gist) as noted in the last line?

evanlitw, 2016-01-25:
Thank you for your response.
I have solved the problem.
However, in Linkit 7688 the use of some of the problems.
I think the issue should be the GPIO.
is it?

usefulthink, 2016-01-25:
I am sorry – I don't quite understand what you mean. 
But I'm glad it somehow worked out. It would be great if you could write a few words on how you solved it for reference. 

But it's very unlikely that this had anything to do with the GPIOs. Those are in no way involved until a program is actually started. And even then, it doesn't use any gpio-related library, but just some fixed functions of the raspberry-processor (which is also why it won't work on other platforms).

closed 2016-02-05 by usefulthink

leonarddevries, 2019-06-07:
> Thank you for your response.
> I have solved the problem.
> However, in Linkit 7688 the use of some of the problems.
> I think the issue should be the GPIO.
> is it?

I'm currently trying to get it to work on OpenWRT as well. @evanlitw Can you maybe ellaborate a little on how you got it to work?