beyondscreen/node-rpi-ws281x-native #47: brightness setting of individual leds

mkpazon, 2016-04-19:


usefulthink, 2016-04-20:
Not sure what this is about as you apparently missed filling in the description text ;)

However I'm going to intepret this as a question and rephrase as "How do I set the brightness for single LEDs?", which is straightforward to explain.

Assuming you have just 3 LEDs, you will have something like this:
```javascript
var ws281x = require('rpi-ws281x-native');
var pixelData = new Uint32Array(3); // create an array for 3 color-values

ws281x.init(3); // initialize the driver for 3 LEDs

// set color-values
pixelData[0] = 0xff0000;
pixelData[1] = 0x00ff00;
pixelData[2] = 0x0000ff;

ws281x.render(pixelData);
```

The color-values are in packed RGB-format (bits 0-7 for blue, 8-15 for green and 16-23 for red, so the value `0xff0000` represents full brightness red, `0x00ff00` full brightness green etc.). The brightness is just part of this: the value per color-channel controls the brightness of the corresponding color-LED (each of the RGB-LEDs contains a single red, green and blue color-LED and relative brightness of these is what mixes the colors).

So, if you just want a very dim red, the color-value could be `0x100000`, or `0x101010` for a dim white.

Hope that helps!

closed 2016-04-20 by usefulthink

mkpazon, 2016-04-20:
Thanks for the quick response!