beyondscreen/node-rpi-ws281x-native #50: installation-problems with node 6

usefulthink, 2016-10-20:
> submitted by @ben0bi as comment in #29

I could not install it with nodejs v6 and I could not install nodejs v4.
For me, it works with nodejs v0.12.

Would like to mark this as todo, compatibility for nodejs >= v4
(Or is it my system? I use the RetroPie-Distro and LXDE as window manager. Multipurpose is the key.)

cernst11, 2016-10-21:
I've been using it in a node v6.4 project of mine for the last couple of months without issue. It's been running on an  rpi0 and the original model B. I can attempt to do a fresh pull of the dependency and have it recompile when I have time to verify.

usefulthink, 2016-10-21:
@cernst11 thanks for the heads up, that's good to know.

ben0bi, 2016-10-23:
+ It's a RasPi 2 B

+ $ curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -
+ apt-get update / apt-get install nodejs
+ npm wsxxx....
+ "Could not be installed. Maybe your nodejs is outdated" or something like this.
+ apt-get purge nodejs
+ $ curl -sL https://deb.nodesource.com/setup_4.x | sudo -E bash -
+ apt-get update / apt-get install nodejs
+ "This version of nodejs does not work with your system." or something alike...
+ $ curl -sL https://deb.nodesource.com/setup_0.12 | sudo -E bash -
+ apt-get update / apt-get install nodejs
+ npm wsxxx...

+ done.

I could not test if it works, but at least it got installed.

usefulthink, 2016-10-24:
thanks, i will have a look.

usefulthink, 2017-12-03:
never heard about this again and it's still running for me with newer node-versions. Please ping to reopen if the problem still exists.

closed 2017-12-03 by usefulthink