beyondscreen/node-rpi-ws281x-native #52: Pwm1

capo349, 2017-02-14:
Hello my name is Carlos I want to know how I can configure pwm to use raspberry channel 1 because I have conflicts with that channel and Audio.

I appreciate your collaboration.

usefulthink, 2017-04-04:
There are several (undocumented and mostly untested) parameters you can pass to the init-function. They will be directly forwarded to the underlying rpi_ws281x-library.

In full, the init-call with default-values looks like this:

```javascript
const ws281x = require('rpi-ws281x-native');

ws281x.init(100, {
  frequency: 800000, // 800kHz (some LEDs run at 400kHz)
  dmaNum: 5, // DMA-number to be used
  gpioPin: 18, // GPIO-PIN to use (this is the only one i tested)
  invert: 0,
  brightness: 255
});
```

So you could experiment especially with the `gpioNum`-value and see if that helps (i recall that there are only two gpios that support using the PWM-modules, but I don't remember which the other one is). If you have success, please report back, so I can add that to the documentation.

stefaanternier, 2017-09-11:
Hi, 
Just like Carlos I have an interest in using your library in combination with  BCM 13 / PWM1 as I want to control two ws281x strings. Trying the default (gpioPin 18) works for me as expected. However when I change the gpioPin to 13, I get your "initialization failed. sorry – no idea why." error.
Any idea why? My code looks as follows:

```
const ws281x = require('rpi-ws281x-native');
ws281x.init(8, {
  gpioPin: 13, 
});
```

usefulthink, 2017-09-11:
Hi @stefaanternier,

the current version doesn't yet support running both PWMs at once. I am currently developing an updated version that will support that (and a lot more...). 
Unfortunately, I do not have the time as I had hoped, but I can publish what I got so far so you could maybe pick it up from there.

stefaanternier, 2017-09-11:
Thanks for your swift response!
That would great. 
Just to be sure. The library by Jeremy Garff offers support for both PWMs? The challenges is here to make this available via your NodeJS wrapper?

usefulthink, 2017-09-11:
Exactly. I didn't try it myself yet (I was always fine with using a single channel) but I am pretty sure others did. But to get this working properly I rewrote the entire thing to move more of the stuff to the JS-part of this module. However, it doesn't work yet and I need to figure out what is going wrong...

usefulthink, 2017-12-01:
see #67 for updates on the new version that will support using both PWM channels.

As we didn't hear back from OP, I am closing this.

closed 2017-12-01 by usefulthink