beyondscreen/node-rpi-ws281x-native #53: Pi Detection

tsuckow, 2017-02-28:
Raspberry Pi 1B not detected:

`$ cat /proc/cpuinfo`

```
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 000d
Serial          : 00000000c5b4676c
```

tsuckow, 2017-02-28:
Changing the regex to:
`              socFamily = cpuInfo.match(/hardware\s*:\s*(bcm2[78][03][589])/i);`
  
And adding:
`              case 'bcm2835': return 1;`

Seems to work...

This is on Arch Linux Arm Pi by the way.

usefulthink, 2017-04-04:
Sorry for not responding earlier, I've been on a longer vacation... 

Your change sounds good to me, maybe open up the regex even more, like `/hardware\s*:\s*(bcm\d+)/i`, as the version-check is happening later anyway.

Can you create a pull-request for this?

cdimascio, 2017-05-13:
@usefulthink , I've prepared a PR - https://github.com/beyondscreen/node-rpi-ws281x-native/pull/58

PR #58 (2017-05-16): Pi Detection 
fixes #53
closed 2017-05-16 by usefulthink