beyondscreen/node-rpi-ws281x-native #55: Cannot use other libraries such as "onoff"

guatedude2, 2017-04-03:
I'm trying to combine this library with https://github.com/fivdi/onoff however once i call `init` I cannot read or write to any other GPIO using onoff.

Any ideas how I can trigger other GPIOs while controlling a neopixel strip?

usefulthink, 2017-04-04:
Honestly, I have no idea. 

I just had a quick look at `onoff` and it seems to be using a gpio kernel-module that makes some/all gpio-functions available to userspace. This module on the other hand requires direct access to some GPIO-related hardware-modules (the PWM-Controllers, DMA and VideoCore memory). This might very well be in conflict with each other although I do not fully understand why.

guatedude2, 2017-04-04:
Yeah i couldnt really pinpoint the issue why it conflicts either. Is there any other library foe the RPi that works along side this one to control other GPIOs? I have yet to find any other library that controls a NeoPixel like this library.

guatedude2, 2017-04-10:
After some searching I found out [rpi-gpio](https://www.npmjs.com/package/rpi-gpio) works with **node-rpi-ws281x-native** library.

Here's an example of the modifies rainbow example:
*I'm using a push button on GPIO24/ PIN 18 of my RPi*

```js
var ws281x = require('rpi-ws281x-native');

var NUM_LEDS = parseInt(process.argv[2], 10) || 10;
var pixelData = new Uint32Array(NUM_LEDS);

ws281x.init(NUM_LEDS);

var gpio = require('rpi-gpio');

gpio.on('change', function(channel, value) {
  console.log('Channel ' + channel + ' value is now ' + value);
  if (value === true) {
    randomColor();
  }
});
gpio.setup(18, gpio.DIR_IN, gpio.EDGE_BOTH);

// ---- trap the SIGINT and reset before exit
process.on('SIGINT', function () {
  ws281x.reset();
  process.nextTick(function () { process.exit(0); });
});

console.log('Press <ctrl>+C to exit.');

var offset = 0;
function randomColor () {
  offset = (Math.random()*1000) % 256;
  for (var i = 0; i < NUM_LEDS; i++) {
    pixelData[i] = colorwheel((offset + i) % 256);
  }

  ws281x.render(pixelData);
}

// rainbow-colors, taken from http://goo.gl/Cs3H0v
function colorwheel(pos) {
  pos = 255 - pos;
  if (pos < 85) { return rgb2Int(255 - pos * 3, 0, pos * 3); }
  else if (pos < 170) { pos -= 85; return rgb2Int(0, pos * 3, 255 - pos * 3); }
  else { pos -= 170; return rgb2Int(pos * 3, 255 - pos * 3, 0); }
}

function rgb2Int(r, g, b) {
  return ((r & 0xff) << 16) + ((g & 0xff) << 8) + (b & 0xff);
}
```

closed 2017-04-10 by guatedude2