beyondscreen/node-rpi-ws281x-native #57: [rpi-ws281x-native] Could not verify raspberry-pi version

divya2100, 2017-05-03:
Hi Team,

I am using raspberry Pi 3 model B.
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000441b8267

But when I load rpi-ws281x-native & run tests, the LED does not change. 
Getting error: [rpi-ws281x-native] Could not verify raspberry-pi version.

Could you please help.

Thanks,
Divya Bansal

jivemoo, 2017-05-05:
I see the same problem on Jessie.

Linux raspberrypi 4.9.25-v7+ #994 SMP Fri Apr 28 16:56:00 BST 2017 armv7l GNU/Linux

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

cdimascio, 2017-05-05:
i see it too

the following code determines the pi version

```
    var raspberryVersion = (function() {
        var cpuInfo = require('fs').readFileSync('/proc/cpuinfo').toString(),
            socFamily = cpuInfo.match(/hardware\s*:\s*(bcm270[89])/i);

        if(!socFamily) { return 0; }

        switch(socFamily[1].toLowerCase()) {
            case 'bcm2708': return 1;
            case 'bcm2709': return 2;
            default: return 0;
        }
    } ());
```

@divya2100 if that returns 0, you get the warning and stubbed bindings. 
you could try removing that check 😨 and it will attempt to create native binding. it may be risky, no idea what that code does... fry the pi?

linpy, 2017-05-08:
I am using raspberry Pi 3 model B on jessie.
And I got Hardware info below from /proc/cpuinfo file. 
==
Hardware : BCM2835

My friend use the same version of pi, she got BCM 2709, she could load rpi-ws281x-native & run tests successfully, I tried to remove the raspberryVersion routine, and LED  change color correctly now. 

I still wonder why the Hardware info different with same hardware. Anyone could provide some clue?

17812, 2017-05-08:
I also have BCM2835 audio profile and getting the same error. I think this happened after recent kernel update...I wonder if there is any way to specify different module?

divya2100, 2017-05-11:
I tried to change the code, but still the same error. 
I added line for BCM2835, but still same error.
Any suggestions for the same.

        var raspberryVersion = (function() {
        var cpuInfo = require('fs').readFileSync('/proc/cpuinfo').toString(),
            socFamily = cpuInfo.match(/hardware\s*:\s*(bcm270[89])/i);

        if(!socFamily) { return 0; }

        switch(socFamily[1].toLowerCase()) {
            case 'bcm2708': return 1;
            case 'bcm2709': return 2;
            default: return 0;

Jkat19, 2017-05-12:
Where are you guys seeing the code to change in order to get the RGB lights to work properly? I cannot find the raspberryVersion routine code on my system. My hardware is also bcm2835.

cdimascio, 2017-05-13:
I've prepared a PR with the fix https://github.com/beyondscreen/node-rpi-ws281x-native/pull/58

victordibia, 2017-05-17:
Thanks @cdimascio!

divya2100, 2017-05-17:
Thanks @victordibia @cdimascio !!!

closed 2017-05-17 by divya2100