beyondscreen/node-rpi-ws281x-native #59: Support newer version of the ws281x library

natcl, 2017-05-15:
The newer version supports more types of leds like RGBW strips. Any chances of using the more recent version ?

usefulthink, 2017-05-16:
I will have some spare time in the coming months and another LED-related project to do, so I will get to that soon. Was on my list anyway :)

paulhayes, 2017-11-29:
Did this never happen?

usefulthink, 2017-11-30:
Unfortunately, I only find so much time to take care of this project but now the new, mostly rewritten version is slowly taking shape, see here: https://github.com/beyondscreen/node-rpi-ws281x-native/tree/v1.x. 

Right now this is just a blind rewrite (meaning the code has yet to see a compiler) as I somehow lost access to the PI I was testing on :/. Something to figure out over the weekend when I hopefully find some time for it.

I'd be happy if someone of you wants to lend me a hand with this. 
The two things most important right now are

 - get the native bindings in [`src/rpi-ws281x.cc`](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/v1.x/src/rpi-ws281x.cc) to compile (well, at least see the error-messages)
 - figure out if the new API is good the way I wrote it now (see [`examples/new-api.js`](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/v1.x/examples/new-api.js) and [`lib/ws281x-native.js`](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/v1.x/lib/ws281x-native.js)

usefulthink, 2017-11-30:
Quick update: the code for the next version just passed my initial test (only seen it on my oscilloscope, but the signal looks correct). There are still a few things that need to be fixed, but we're getting there.

If you feel adventurous: check out the v1.x branch, run `npm install` on a PI and adjust/run the example: `sudo node examples/new-api.js`.
Please tell me about any errors you might see.

usefulthink, 2017-11-30:
See #67

paulhayes, 2017-12-01:
Thanks @usefulthink, I'm really glad you are doing this. I forked that branch soon after posting to this issue, and began in earnest trying to pick through things. Thanks for the intro on #67, that will help me. I'll take a look at the new api, and see if I can get my RGBW strand ( 2 meters of 160p/m ) to work.

natcl, 2018-08-15:
If you use the 1.x branch it should work.