beyondscreen/node-rpi-ws281x-native #60: Support for Pi 3 Model B?

cpottamus, 2017-06-09:
I'm having a very difficult time getting this library to work with Pi 3 Model B.

Using an 74AHCT125 level converter on my breakout board, similar to this [guide for the ws281x python library](https://learn.adafruit.com/neopixels-on-raspberry-pi/)

I'm using NeoPixels but I can't get any voltage out of my converter... wondering if there is a library support issue.

I've already modified my audio param and HDMI hotplugs, per the [pi 3 support issue in ws281x](https://github.com/jgarff/rpi_ws281x/issues/103).

Let me know if you have any thoughts, or if this library simply isn't up to date enough to support pi 3.

Thanks!

shortbloke, 2017-06-13:
@cpottamus I've just built this using a RPi3 and 74HCT125 level converter. If you're trying to measure voltage via multi-meter you won't be able to as the data commands a very short pulses. You'd need an oscilloscope to be able to see it. 

If you are powering a long strip of lights you'll need a large PSU, I'm using a 5V 10A power supply to power both Pi and my LEDs, if using different supplies for your Pi and your LEDs, you'll need to ensure they have a common ground. 

I had issues getting my 144 1m strip working reliably and not just flickering all over the place. Firstly 1 had to connect a 470 ohm resistor from the 74HCT125 output (pin 3) to the data input line for my strip.

I also had issues depending on what (tv vs monitor vs nothing) was connected to the HDMI port, and found that editing `/boot/config.txt` and commenting out `dtparam=audio=on` looks to have fixed it. 

Hope this helps, but I can confirm the library does work on an RPi3.

usefulthink, 2017-07-04:
Hi! Yes, the library got a little bit dated and I had some problems myself (but on a PI2) when I started a new project with it. Based on that and some other feedback here, I started a almost complete rewrite based on the latest Version of the upstream-library and some new design-principles for the c++-bindings which is now well underway. So there will be something worth testing within the next few days.

LolCatPhd, 2017-07-08:
I need some help please:
I have a ws2812 neopixel (8 strip) hooked up to a RPi3 B using 74HCT125 connected in between GPIO18 and the neopixel strip. The 74HCT125 does output 5V when provided by a 3v3 constant voltage. I have commented out dtparam=audio=on and blacklisted the following:

blacklist snd_bcm2835
blacklist snd_pcm
blacklist snd_timer
blacklist snd_pcsp
blacklist snd

I have tried both the ws281x python library as well as this: https://flows.nodered.org/node/node-red-node-pi-neopixel

When I start strandtest.py or send colour commands with node-red i hear the noises coming through speakers still. Is that normal?

My strip remains completely unresponsive, no lights are coming on....

What should the output of the command "lsmod" be?

"lsmod" gives me this:

meaku, 2017-07-10:
@usefulthink great to hear you are working on a new version. Let me know if there is anything to play with / test.

Granjow, 2017-11-04:
@cpottamus Works fine here. Did you use the correct GPIO pin (Pin 12 or BCM 18)? https://pinout.xyz/

Something like a [Picoscope](https://www.picotech.com/oscilloscope/2000/picoscope-2000-overview) or [Saleae](https://www.saleae.com/) might help too.

mikaye, 2017-11-20:
@usefulthink How long until a new version with the latest rpi-ws281x? I'm currently getting the error that it thinks the RPi3 is not a Pi. According to @jgarff, this should be fixed with the latest rpi-ws281x.

usefulthink, 2017-11-30:
just passed the first tests, see #67