beyondscreen/node-rpi-ws281x-native #61: File System going read-only! With latest raspbian updates

shortbloke, 2017-06-17:
I originally raised this issue against another project: https://github.com/jamesbulpin/meshblu-connector-ws2811/issues/4 However after some further investigation it appear the issue is rooted in this library.

Issue:
After running the test program or the examples on a Raspberry Pi 3 with Raspbian Jessie + all updates as of 17 June 2017. The file system gets marked read-only. Checking dmesg shows lots of errors from the moment the program starts:
```
[  303.215569] mmc0: timeout waiting for hardware interrupt.
[  303.218188] mmcblk0: error -110 transferring data, sector 10372992, nr 8, cmd response 0x900, card status 0xc00
[  313.455603] mmc0: timeout waiting for hardware interrupt.
[  313.465817] mmcblk0: error -110 transferring data, sector 3767312, nr 8, cmd response 0x900, card status 0xc00
[  323.695659] mmc0: timeout waiting for hardware interrupt.
[  323.698214] mmcblk0: error -110 transferring data, sector 3767312, nr 8, cmd response 0x900, card status 0xc00
[  323.698359] blk_update_request: I/O error, dev mmcblk0, sector 3767312
[  323.698418] Aborting journal on device mmcblk0p2-8.
[  333.935677] mmc0: timeout waiting for hardware interrupt.
[  333.938401] mmcblk0: error -110 transferring data, sector 92160, nr 8, cmd response 0x900, card status 0xc00
[  344.175717] mmc0: timeout waiting for hardware interrupt.
[  344.178360] mmcblk0: error -110 transferring data, sector 92160, nr 8, cmd response 0x900, card status 0xc00
[  344.178480] blk_update_request: I/O error, dev mmcblk0, sector 92160
[  344.178502] Buffer I/O error on dev mmcblk0p2, logical block 0, lost sync page write
[  344.178552] EXT4-fs (mmcblk0p2): previous I/O error to superblock detected
[  344.179720] EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:56: Detected aborted journal
[  344.179727] EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:56: Detected aborted journal
[  344.179738] EXT4-fs (mmcblk0p2): Remounting filesystem read-only
[  344.180900] EXT4-fs (mmcblk0p2): ext4_writepages: jbd2_start: 6144 pages, ino 22151; err -30
```
See attached
[dmesg.txt](https://github.com/beyondscreen/node-rpi-ws281x-native/files/1082656/dmesg.txt) for full log.

I've reproduced this on 3 different (all new) SD Cards:
* Sandisk 8GB - man:0x000003 oem:0x5344 name:SL08G hwrev:0x8 fwrev:0x0
* Samsung Pro 16GB - man:0x00001b oem:0x534d name:00000 hwrev:0x1 fwrev:0x0
* Integral 16GB - man:0x00009e oem:0x4245 name: hwrev:0x1 fwrev:0x0

Everything works fine when using the base Raspbian Image (2017-04-10-raspbian-jessie), it breaks with updates issued after that release. These updates are:
```
PROGRAM: login INSTALLED: 1:4.2-3+deb8u3 AVAILABLE: 1:4.2-3+deb8u4
PROGRAM: perl INSTALLED: 5.20.2-3+deb8u6 AVAILABLE: 5.20.2-3+deb8u7
PROGRAM: perl-base INSTALLED: 5.20.2-3+deb8u6 AVAILABLE: 5.20.2-3+deb8u7
PROGRAM: perl-modules INSTALLED: 5.20.2-3+deb8u6 AVAILABLE: 5.20.2-3+deb8u7
PROGRAM: libc6-dev INSTALLED: 2.19-18+deb8u7 AVAILABLE: 2.19-18+deb8u9
PROGRAM: libc-dev-bin INSTALLED: 2.19-18+deb8u7 AVAILABLE: 2.19-18+deb8u9
PROGRAM: rpcbind INSTALLED: 0.2.1-6+deb8u1 AVAILABLE: 0.2.1-6+deb8u2
PROGRAM: libc-bin INSTALLED: 2.19-18+deb8u7 AVAILABLE: 2.19-18+deb8u9
PROGRAM: libc6-dbg INSTALLED: 2.19-18+deb8u7 AVAILABLE: 2.19-18+deb8u9
PROGRAM: libc6 INSTALLED: 2.19-18+deb8u7 AVAILABLE: 2.19-18+deb8u9
PROGRAM: libtirpc1 INSTALLED: 0.2.5-1 AVAILABLE: 0.2.5-1+deb8u1
PROGRAM: libudev1 INSTALLED: 215-17+deb8u6 AVAILABLE: 215-17+deb8u7
PROGRAM: udev INSTALLED: 215-17+deb8u6 AVAILABLE: 215-17+deb8u7
PROGRAM: initramfs-tools INSTALLED: 0.120+deb8u2 AVAILABLE: 0.120+deb8u3
PROGRAM: libsystemd0 INSTALLED: 215-17+deb8u6 AVAILABLE: 215-17+deb8u7
PROGRAM: libpam-systemd INSTALLED: 215-17+deb8u6 AVAILABLE: 215-17+deb8u7
PROGRAM: systemd INSTALLED: 215-17+deb8u6 AVAILABLE: 215-17+deb8u7
PROGRAM: systemd-sysv INSTALLED: 215-17+deb8u6 AVAILABLE: 215-17+deb8u7
PROGRAM: libgnutls-openssl27 INSTALLED: 3.3.8-6+deb8u4 AVAILABLE: 3.3.8-6+deb8u5
PROGRAM: libtasn1-6 INSTALLED: 4.2-3+deb8u2 AVAILABLE: 4.2-3+deb8u3
PROGRAM: libgnutls-deb0-28 INSTALLED: 3.3.8-6+deb8u4 AVAILABLE: 3.3.8-6+deb8u5
PROGRAM: libldap-2.4-2 INSTALLED: 2.4.40+dfsg-1 AVAILABLE: 2.4.40+dfsg-1+deb8u2
PROGRAM: libavutil54 INSTALLED: 6:11.8-1~deb8u1+rpi1 AVAILABLE: 6:11.9-1~deb8u1+rpi1
PROGRAM: libavresample2 INSTALLED: 6:11.8-1~deb8u1+rpi1 AVAILABLE: 6:11.9-1~deb8u1+rpi1
PROGRAM: libavcodec56 INSTALLED: 6:11.8-1~deb8u1+rpi1 AVAILABLE: 6:11.9-1~deb8u1+rpi1
PROGRAM: librtmp1 INSTALLED: 2.4+20150115.gita107cef-1 AVAILABLE: 2.4+20150115.gita107cef-1+deb8u1
PROGRAM: libavformat56 INSTALLED: 6:11.8-1~deb8u1+rpi1 AVAILABLE: 6:11.9-1~deb8u1+rpi1
PROGRAM: libisc-export95 INSTALLED: 1:9.9.5.dfsg-9+deb8u10 AVAILABLE: 1:9.9.5.dfsg-9+deb8u11
PROGRAM: libdns-export100 INSTALLED: 1:9.9.5.dfsg-9+deb8u10 AVAILABLE: 1:9.9.5.dfsg-9+deb8u11
PROGRAM: libgudev-1.0-0 INSTALLED: 215-17+deb8u6 AVAILABLE: 215-17+deb8u7
PROGRAM: libicu52 INSTALLED: 52.1-8+deb8u4 AVAILABLE: 52.1-8+deb8u5
PROGRAM: libisccfg-export90 INSTALLED: 1:9.9.5.dfsg-9+deb8u10 AVAILABLE: 1:9.9.5.dfsg-9+deb8u11
PROGRAM: libirs-export91 INSTALLED: 1:9.9.5.dfsg-9+deb8u10 AVAILABLE: 1:9.9.5.dfsg-9+deb8u11
PROGRAM: libnss3 INSTALLED: 2:3.26-1+debu8u1 AVAILABLE: 2:3.26-1+debu8u2
PROGRAM: libwbclient0 INSTALLED: 2:4.2.14+dfsg-0+deb8u5 AVAILABLE: 2:4.2.14+dfsg-0+deb8u6
[... 25 lines cut ...]
PROGRAM: libblas-common INSTALLED: 1.2.20110419-10 AVAILABLE: 3.7.0-1~bpo8+1
PROGRAM: libblas3 INSTALLED: 1.2.20110419-10 AVAILABLE: 3.7.0-1~bpo8+1
PROGRAM: libbluray1 INSTALLED: 1:0.6.2-1 AVAILABLE: 1:0.9.3-3
PROGRAM: libcupsfilters1 INSTALLED: 1.0.61-5+deb8u3 AVAILABLE: 1.0.61-5+deb8u3+rpi1
PROGRAM: liblapack3 INSTALLED: 3.5.0-4 AVAILABLE: 3.7.0-1~bpo8+1
PROGRAM: libmwaw-0.3-3 INSTALLED: 0.3.1-2 AVAILABLE: 0.3.1-2+deb8u1
PROGRAM: raspberrypi-sys-mods INSTALLED: 20170313 AVAILABLE: 20170519
PROGRAM: raspi-config INSTALLED: 20170307 AVAILABLE: 20170503
PROGRAM: rc-gui INSTALLED: 1.7 AVAILABLE: 1.9
PROGRAM: realvnc-vnc-server INSTALLED: 6.0.3.27095 AVAILABLE: 6.1.1.28093
PROGRAM: realvnc-vnc-viewer INSTALLED: 6.0.3.27095 AVAILABLE: 6.1.1.28093
PROGRAM: rpi-chromium-mods INSTALLED: 20170317 AVAILABLE: 20170614
PROGRAM: sudo INSTALLED: 1.8.10p3-1+deb8u3 AVAILABLE: 1.8.10p3-1+deb8u4
PROGRAM: unzip INSTALLED: 6.0-16+deb8u2 AVAILABLE: 6.0-16+deb8u3
PROGRAM: libraspberrypi-dev INSTALLED: 1.20170405-1 AVAILABLE: 1.20170427-1
PROGRAM: libraspberrypi-doc INSTALLED: 1.20170405-1 AVAILABLE: 1.20170427-1
PROGRAM: raspberrypi-kernel INSTALLED: 1.20170405-1 AVAILABLE: 1.20170427-1
PROGRAM: libraspberrypi-bin INSTALLED: 1.20170405-1 AVAILABLE: 1.20170427-1
PROGRAM: libraspberrypi0 INSTALLED: 1.20170405-1 AVAILABLE: 1.20170427-1
PROGRAM: raspberrypi-bootloader INSTALLED: 1.20170405-1 AVAILABLE: 1.20170427-1
PROGRAM: raspberrypi-ui-mods INSTALLED: 1.20170315 AVAILABLE: 1.20170503
```
I've not narrowed down which update triggers this conflict. I can confirm that running other applications including sysbench tests of the  filesystem and memory don't trigger any problems.

Once the system has gone read-only it recovers if you reboot the RPi.

usefulthink, 2017-07-04:
Oh well, I probably need to get a PI3 for testing at some point.. I'm sorry you are experiencing these problems :( And thank you very much for this very detailed report. 

Now, this module in itself doesn't do anything fancy. It's just a pretty slim wrapper around the [original library by jgarff](https://github.com/jgarff/rpi_ws281x). So, my first guess would be that it has something to do with that library.

Could you maybe check if you get the same problems when running the example from that library?

Otherwise (me starting wild guessing), one possible point of intersection with anything SDCard-related could be the use of DMA for data-transfer. Someone on that other thread said that changing the dma-channel to 10 did help. You can try that as well with this library:

```javascript
var ws281x = require('ws281x');

ws281x.init(100, {dmaNum: 10}); // default value for dmaNum is 5
```
(Note: I never tested these paramters myself, but it should work)

The default DMA-Channel used is 5 but unfortunately I don't have any good Idea about how to debug this. I searched around a bit on my PI2 and found some dma-related information in the sysfs-tree at `/sys/devices/platform/soc/3f007000.dma/dma` but wasn't able to get any useful information from there... I guess I just don't know enough about how that works to be really helpful here.

shortbloke, 2017-07-05:
Yes, this bug appears to be in the base library pulled into this project. The using DMA channel 10 didn't work for the original reporter, but they did say they had issues with lights flickering, they didn't mention if they were getting the SD card errors still. I'll give it a try in the next day or so and update here and the other issue against the base library.

shortbloke, 2017-07-06:
I've tested with the latest version of jgarff's library and can confirm if I use the default DMA of 5 I have the problem, if I switch to DMA 10 then it's fixed. 
I've repeated a test by modifying the `example\rainbow.js` to use DMA 10:
```
ws281x.init(NUM_LEDS, {dmaNum: 10});
```
This appears to also address the issue. I expect the fix to possibly change the default DMA number would be best done in the base project. 
However, maybe there is something that can be done here to make changing the DMA easier for the examples and core implementation.

usefulthink, 2017-07-06:
Awesome, glad to hear that! I will at least make it more obvious and add that to the documentation. 

Can you append the output of `cat /proc/cpuinfo`, `cat /etc/os-release` and `uname -a` here for reference? I may be able to implement an automatic switch based on hardware- and kernel-version.

shortbloke, 2017-07-06:
Sure here's the info:
**cpuinfo**
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000ff564eda
```

**os-release**
```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

**uname**
```
Linux raspberrypi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux
```

jsonpoindexter, 2017-08-13:
I am having the same issues! Glade I found this thread!

chilipeppr, 2017-08-21:
I found a similar issue with the SD card switching to read-only mode and from what I can tell switching to DMA 10 fixed it. I'm on a Raspberry Pi 3. 

For what it's worth, I couldn't find docs on the options either that you can pass in. Here's what I deduced.

var OPTIONS = {
  // frequency: 8000,  // Uint32Value symFreq = Nan::New<String>("frequency").ToLocalChecked(),
  dmaNum : 10,      // Int32Value symDmaNum = Nan::New<String>("dmaNum").ToLocalChecked(),
  gpioPin: 18,    // Int32Value symGpioPin = Nan::New<String>("gpioPin").ToLocalChecked(),
  // invert: 0,     // Int32Value symInvert = Nan::New<String>("invert").ToLocalChecked(),
  brightness: 255, // Int32Value symBrightness = Nan::New<String>("brightness").ToLocalChecked();
}
ws281x.init(NUM_LEDS, OPTIONS);

etx, 2017-11-17:
I think this fixed the problem I was having on a RPi Zero w running this lib. DMA 10 seems more stable and i'm not seeing the lockups I was having before. It's been killing me all week! 

Great lib, super helpful!

d74g0n, 2018-07-06:
It is true about DMA fixing my problems;  
ws281x.init(NUM_LEDS,{dmaNum: 10}); // simply adding dmaNum: 10 has stabilized my system that has now been running for a month+ without problems.  The DMA conflict not only caused read only errors to occur but really stuttered and lagged all samba and other filesystem functions before catastrophic read only error occurred.   On a side note i fried a pi when the 12v LED line shorted thru the pi...  it did not fry the SD card thank goodness.