beyondscreen/node-rpi-ws281x-native #63: Compatibility with Johnny Five?

gangstead, 2017-07-30:
Has anyone gotten this to work with Johnny five?  I think the native library and rasp-io don't get along even though they are going after different pins.

I made some test code to take the working rainbow example and speed it up while a button is being held down:
```js
const ws281x = require('rpi-ws281x-native');
const five = require('johnny-five');
const Raspi = require('raspi-io');

const board = new five.Board({
  io: new Raspi()
});

let rainbow;

const NUM_LEDS = parseInt(process.argv[2], 10) || 20,
    pixelData = new Uint32Array(NUM_LEDS);

ws281x.init(NUM_LEDS);

// ---- trap the SIGINT and reset before exit
process.on('SIGINT', () => {
  ws281x.reset();
  process.nextTick(() => { process.exit(0); });
});

const RAINBOW_FAST = 100;
const RAINBOW_SLOW = 1000;
let offset = 0;

const makeRainbow = (speed) => {
  if (rainbow) {
    clearTimeout(rainbow);
  }

  rainbow = setInterval(() => {
    for (let i = 0; i < NUM_LEDS; i++) {
      pixelData[i] = colorwheel((offset + i) % 256);
    }

    offset = (offset + 1) % 256;
    ws281x.render(pixelData);
  }, speed / 30);
}

console.log('Press <ctrl>+C to exit.');

// rainbow-colors, taken from http://goo.gl/Cs3H0v
function colorwheel(pos) {
  pos = 255 - pos;
  if (pos < 85) { return rgb2Int(255 - pos * 3, 0, pos * 3); }
  else if (pos < 170) { pos -= 85; return rgb2Int(0, pos * 3, 255 - pos * 3); }
  else { pos -= 170; return rgb2Int(pos * 3, 255 - pos * 3, 0); }
}

function rgb2Int(r, g, b) {
  return ((r & 0xff) << 16) + ((g & 0xff) << 8) + (b & 0xff);
}

board.on('ready', function() {
  const button = new five.Button('GPIO23');
  button.on('down', () => {
    console.log('button down');
    makeRainbow(RAINBOW_FAST);
  });

  button.on('up', () => {
    console.log('button up');
    makeRainbow(RAINBOW_SLOW);
  });
});

makeRainbow('RAINBOW_SLOW');

// ---- animation-loop
var offset = 0;
setInterval(function () {
  for (var i = 0; i < NUM_LEDS; i++) {
    pixelData[i] = colorwheel((offset + i) % 256);
  }

  offset = (offset + 1) % 256;
  ws281x.render(pixelData);
}, 1000 / 30);
```

*What I expect to happen:* Rainbow effect speeds up when button is pressed, slows down when released.

*What actually happens:* I get the button up/down console logs from the johnny-five part, but the rainbow doesn't display.  The LEDs don't light at all.

usefulthink, 2017-12-03:
Sorry for not getting to this earlier, but I suspect that the J5-library itself is somehow interfering with direct configuration and access of the DMA/PWM/GPIO functions of the raspberry.

One thing that comes to mind is you can try to make the GPIO usage more exlicit by specifying the includePins or excludePins options to the `Raspi` constructor. That way it might be possible to prevent the interference of the two libraries.

joakimunge, 2018-08-27:
I can confirm excluding the pin used by the LEDs will make it available and working. Tried just now with Raspberry Pi 3 B+ and NeoPixel Stick. In the default case, that is GPIO18.

```js
const ws281x = require('rpi-ws281x-native');
const five = require('johnny-five');
const Raspi = require('raspi-io');

const board = new five.Board({
  io: new Raspi({ excludePins: 'GPIO18'})
});
```

I think we can close this?

usefulthink, 2018-08-27:
sounds great! Thanks for updating this. Will close it now. @gangstead, if you still experience problems, feel free to reopen or create a new issue.

closed 2018-08-27 by usefulthink

christopher4lis, 2021-02-02:
Just tested this on my end—can't seem to get it working even with the excluded pins:

```
import five from 'johnny-five'
import ws281x from 'rpi-ws281x-native'
import { RaspiIO } from 'raspi-io'
import { hslToHex } from './lib.js'

const board = new five.Board({
  io: new RaspiIO({
    excludePins: 'GPIO18'
  })
})

const NUM_LEDS = 144
const pixelData = new Uint32Array(NUM_LEDS)

// must be initialized after board is setup, otherwise, johnny-five doesn't know with pin to "exclude," or use as the LED pin
ws281x.init(NUM_LEDS)

for (let i = 0; i < NUM_LEDS; i++) {
  pixelData[i] = 0xffffff
}
ws281x.render(pixelData)
```

Bottom, library code works as expected stand-alone, but not with the board initialization. Any ideas?