beyondscreen/node-rpi-ws281x-native #64: [rpi-ws281x-native] it looks like you are not running on a raspberry pi

socalledsound, 2017-09-08:
Hello, I went through the same Adafruit tutorial that you linked to with a single neopixel on a raspberry pi 2 and am able to get a rainbow animation using their python code.  I'd prefer to work in JS so thought I'd try out your library.  Unfortunately when I run sudo node rainbow.js I get:

[rpi-ws281x-native] it looks like you are not running on a raspberry pi so there will be no functionality exposed by this module. For convenience, implementation stubs are provided.

Any ideas why this is happening?  Help appreciated!

socalledsound, 2017-09-08:
ps: is 'linux' in the pi detection code case senstive?  when I run uname -a I get 

Linux raspberrypi 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

b33son, 2017-10-07:
Just FYI, this library works on Raspberry Pi Zero W. I've successfully got it working with OPC and OLA.

Granjow, 2017-11-04:
@socalledsound it checks this:

    if (process.arch !== 'arm' && process.platform !== 'linux')

Do those differ for you?

usefulthink, 2017-11-30:
There are quite some raspberry-editions that might give false negatives with the current detection-code. The upcoming version will use completely rewritten code for that.

paulhayes, 2017-12-02:
Hi there, I discovered this problem too, the rpi-version library hasn't been updated in two years! I've added the RpiZeroW cpu to the list of cpu info in the library and performed a pull request. Hopefully the maintainer will merge in the change soon!

https://github.com/ivsgroup/rpi-version/pull/4

usefulthink, 2017-12-03:
@paulhayes Awesome, that did at least already land on master. Just needs a new version on npm now.

closed 2018-07-05 by usefulthink