beyondscreen/node-rpi-ws281x-native #65: Using any control line other than BCM 18

scottstratford, 2017-10-31:
Has anyone gotten any other interface to work? I am doing this:

```
   var ws2812 = require('../../node_modules/rpi-ws281x-native/lib/ws281x-native');
   ws2812.init(NUM_LEDS, {gpioPin: 18});

```
and if I use any value other than 18 (i.e. 10 or 21) as the gpioNum I get this error:

```
   <snip>/node_modules/rpi-ws281x-native/lib/ws281x-native.js:136
   bindings.init(numLeds, options);
   ^
   Error: init(): initialization failed. sorry – no idea why.
   at Error (native)
```

I am running on a Raspberry Pi Zero. Pin 18 is basically working but I am getting irregular delays on the write command. I am driving a 60 LED strip and most calls to write take <2ms, but every minute or so a write will take ~600ms to ~1200ms ... I'm only looking to use SPI to see if this problem can be solved.

usefulthink, 2017-11-30:
I'm closing this as this will be resolved with version 1.0 (now finally on the horizon). See #67 for progress on that.

closed 2017-11-30 by usefulthink

hardillb, 2018-03-14:
This still doesn't work with the node-rpi-ws281x-native@next builds

```
const pixel = 13;
const neopixel = ws281x(1,{gpio: pixel, stripType: 0x00081000});
```

Throws the following error:

```
Error: Selected GPIO not possible
    at Error (native)
    at init (/home/pi/N556/node_modules/rpi-ws281x-native/lib/ws281x-native.js:156:12)
    ...
```

Any idea when we might be able to use pin 13?

hardillb, 2018-03-14:
Ignore that, I found the new README and things look better.

usefulthink, 2018-03-14:
hm, hard to say. That is an error that is forwarded from the underlying c-library which now supports multiple operating modes (PWM and PCM) to support as many pins as possible. There is some more information about this in [that projects readme](https://github.com/jgarff/rpi_ws281x#gpio-usage)