beyondscreen/node-rpi-ws281x-native #67: v1.0 consolidation

usefulthink, 2017-11-30:
A rewrite of the library with adjusted API and rewritten native bindings is on it's way and did just pass some initial testing.

The new code can be found here: https://github.com/beyondscreen/node-rpi-ws281x-native/tree/v1.x

This issue is used to consolidate all open issues that are resolved by these changes.

paulhayes, 2017-12-02:
I'm so happy, with a bit of tweaking the new-api.js worked great for me!
I might do a pull request, but I'm so glad it's working with me rgbw LED strip!

ps I noticed you have an ar-drone project on github from ages ago, me too!

paulhayes, 2017-12-02:
FYI rather than the npm upload. I use sshfs which I find to be a great way of using GUI editor for code on my pi!

usefulthink, 2017-12-03:
Updated the [README](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/v1.x/README.md) with the new API. Please let me know if you like it or if there are any things I can improve in the API-Design (that will likely stay for the foreseeable future...).

shaunjacobsen, 2018-01-17:
Hi! I am working on a project and would like to use this library with an RGBW ring. Will the 1.0 version be available soon on npm (e.g. using npm install rpi-ws281x-native@1.0)?

usefulthink, 2018-01-17:
Right now it is available for testing via `npm install rpi-ws281x-native@next` ([docs here](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/v1.x/README.md)). 

There may be some things that do not fully work yet. If you encounter any problems please be so kind to report back here :)

ecdeveloper, 2019-11-21:
How are the things going with this rc? Any chance we get that merged/released any time soon?