beyondscreen/node-rpi-ws281x-native #70: setBrightness 0 - 100 not 0 - 255

XPilot, 2017-12-17:
I've metered the setBrightness function and it looks like the scale is from 0 to 100 not 0 - 255

natcl, 2018-03-14:
Any updates on this ? I'd be curious to know too.

XPilot, 2018-03-27:
@natcl - Apparently not. We will have to use scale 0-100 instead of 0-255.

JeffJassky, 2019-09-10:
Any update on this?

usefulthink, 2019-09-10:
Not sure what you're asking for here. 
The brightness-value as specified is directly passed on to the c-library that uses it to scale the rgb-values.

https://github.com/beyondscreen/node-rpi-ws281x-native/blob/28bc61d19c3f92d044be9b79628fa8cbd4abec46/lib/ws281x-native.js#L180-L188

https://github.com/beyondscreen/node-rpi-ws281x-native/blob/28bc61d19c3f92d044be9b79628fa8cbd4abec46/src/rpi-ws281x.cc#L134-L147

and finally:

https://github.com/beyondscreen/rpi_ws281x/blob/05248980e760c43fe87ad905ae5f6edf8b824560/ws2811.c#L628-L633

JeffJassky, 2019-09-10:
I see. I don't know C well but looking at the `scale` declaration it seems like it's 0-100.

Note that the README `https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/README.md` says `* @param {Number} brightness the brightness, value from 0 to 255.`. Maybe just the README needs to be updated.

usefulthink, 2019-09-11:
you mean `int scale   = (channel->brightness & 0xff) + 1;`? this takes the brightness-value, sets all but the lowest 8 bits to zero and adds 1, resulting in a value-range from 1 to 256.

JeffJassky, 2019-09-11:
*facepalm* my bad. Thank you for clarifying. I just tried that out in a C interpreter and realize how it works.

`setBrightness()` takes a value from 0-255.

Thanks, @usefulthink