beyondscreen/node-rpi-ws281x-native #73: Lower values only update every 16

JasXSL, 2018-03-01:
I'm running an Adafruit Neopixel strip of 120 lights (Should be ws2812), hooked up to a raspberry pi via a 74AHCT125 3.3v to 5v converter. Values on the darker ranges only seem to update every 16 steps. For an example:

- 0x000009 = Completely off
- 0x000010 = On
- 0x000019 = Same as 0x000010
- 0x000020 = Very notably brighter than 0x000019

Is this an issue of hardware, or am I doing something wrong?

JasXSL, 2018-03-01:
I've tested without the 3.3v to 5v converter and I'm getting the same issue. Only every 0x10 will actually update the brightness.

usefulthink, 2018-03-14:
in the pre-1.0 version there is gamma-correction in place that might be responsible for this: https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/lib/ws281x-native.js#L74-L89