beyondscreen/node-rpi-ws281x-native #76: Does not run on Raspberry pi 3 B+

natcl, 2018-03-29:
Tried the v1.0.0 version on Raspberry pi 3 B+ and it says it could not verify the raspberry version and to file a bug report.

Thanks !

Thyvo, 2018-04-10:
Hi natcl, I would like to use your fork of this libary but I receive an build error while making it from source.

Could you activate the issues tab so I can post the error?

natcl, 2018-04-10:
Should work now

tsuckow, 2018-06-13:
I also published rpi-version-next@1.5.0 with your changes.

alberto2000, 2018-08-27:
Can we merge B+ support into v1.x?

usefulthink, 2018-08-27:
thanks for bringing this up, I will have a look tonight!

joakimunge, 2018-08-27:
@usefulthink Nice. I'm working on an IoT project using neopixels and RPi 3 B+ right now, so support would be great.

@natcl I tried to build your repo but I received an error, probably the same as @Thyvo . Posted an issue in your fork.

alberto2000, 2018-08-27:
I got it to work by adding `"a020d3" : "Model_B+_PI_3"` to index.js of rpi-version and updating/cloning current master branch of https://github.com/jgarff/rpi_ws281x into ./src/rpi_ws281x and rerunning `npm i`

joakimunge, 2018-08-27:
@alberto2000 Was this using @natcl fork or the source v1.x?

alberto2000, 2018-08-27:
@joakimunge I'm using v1.x

joakimunge, 2018-08-27:
@alberto2000 Thanks. Got it to work using your way as well.

alberto2000, 2018-08-29:
The things needed to to to make this work with B+ are
- update submodule at `./src/rpi_ws281x` to use official/latest jgarff version
- use git version of `rpi-version` (npm package is out of date)

joakimunge, 2018-08-30:
Can confirm. @alberto2000

tjgionet, 2019-02-24:
@alberto2000 I'm still having trouble getting this to run even after both your updates listed above. 
There is literally no response from either the neopixels or out the console. 

Any other tips? 
When you update the submodule rpi_ws281x and run 'npm i' is there anyting else that needs to be done.  

Basically I do both updates above head to the root of the directory complete a final npm i then a node-gyp rebuild. Nothing fails I can init the leds fine and send a render to them but nothing...

I have been able to use the master branch of this repo which works sometimes. 

Thanks!

tsuckow, 2019-02-24:
Are you trying to drive the neopixels directly or do you have a 5v buffer
between the pi and the pixels?

tjgionet, 2019-02-26:
I believe we drive them directly. My issue was simply that I was using the old API. Once I updated that all went smoothly. Thanks!!

riccardolardi, 2019-12-17:
Model a52082 is not in rpi-version

riccardolardi, 2019-12-19:
rpi-version updated with https://github.com/ivsgroup/rpi-version/pull/9