beyondscreen/node-rpi-ws281x-native #77: problems with reset() and finalize() in the new api

natcl, 2018-03-30:
With the new version (1.x)
I get the following error when I call reset():
```
Press <ctrl>+C to exit.
^C/home/pi/leds/node_modules/rpi-ws281x-native/lib/ws281x-native.js:93
    this.array.fill(0);
              ^

TypeError: Cannot read property 'fill' of null
    at Channel.[_reset] (/home/pi/leds/node_modules/rpi-ws281x-native/lib/ws281x-native.js:93:15)
    at channels.forEach.channel (/home/pi/leds/node_modules/rpi-ws281x-native/lib/ws281x-native.js:173:46)
    at Array.forEach (native)
    at Function.reset (/home/pi/leds/node_modules/rpi-ws281x-native/lib/ws281x-native.js:173:12)
    at process.<anonymous> (/home/pi/leds/node_modules/rpi-ws281x-native/examples/new-api.js:10:10)
    at emitNone (events.js:86:13)
    at process.emit (events.js:185:7)
    at Signal.wrap.onsignal (internal/process.js:217:44)
```

Also finalize() does a segmentation fault.

Any ideas ?

Thanks !

natcl, 2018-03-30:
I fixed the first problem by changing this:
```
  [_reset]() {
    if (this.array) this.array.fill(0);
  }
```
The segmentation fault remains upon finalize though.

alberto2000, 2018-09-13:
Same problem here, .finalize() leads to segmentation fault, but weirdly only sometimes