beyondscreen/node-rpi-ws281x-native #82: Installation fails on current node-version

neon-JS, 2018-06-03:
I reinstalled Raspbian and installed the current node (v10.3.0 with npm v6.1.0). 
Installation of `rpi-ws281x-native` failed during `node-gyp rebuild`. 
Using nvm to use node v8.0.0 with npm v5.0.0 helped since it could be build now. 

Logs (v10.3.0, npm v6.1.0):
[2018-06-03T10_11_26_415Z-debug.log](https://github.com/beyondscreen/node-rpi-ws281x-native/files/2065851/2018-06-03T10_11_26_415Z-debug.log)
[terminal-output.txt](https://github.com/beyondscreen/node-rpi-ws281x-native/files/2065852/terminal-output.txt)

I'm definitely not a C/C++ programmer but I think I'll may further investigate this out of curiosity. But I'm not sure if I can help you with this. :/

jonas27, 2018-06-08:
Same problem here. Updating node-gyp to v3.7.0 didn't help either

jonas27, 2018-06-09:
Got it to work. Used Raspberian Jessie, Stable node version for arm7 and version 0.8.2 of this lib

neon-JS, 2018-06-09:
Yes, I got it to work with node 8.0.0 as mentioned above. But building should work on current version too since switching to an old version of node when using this module is neither comfortable, nor safe, I guess.

usefulthink, 2018-07-05:
I'll have to look into this, but unfortunately I don't have a working setup around right now. If anyone wants to take it, I would suspect an outdated version of [nan](https://github.com/nodejs/nan) is the reason for this. Running `npm install nan@latest` and rebuilding might already solve that.

VlivesRaw, 2018-07-31:
This issue seems to be solved with the #79 
@usefulthink can you please publish a new version to npm?

usefulthink, 2018-07-31:
I just pushed out version 0.9.0 with all of these fixes. Probably the last version for the "old" 0.x branch. As I still don't have any working setup around, I will fully rely on this issue-tracker to see if something did go undetected.

closed 2018-07-31 by usefulthink