beyondscreen/node-rpi-ws281x-native #85: Cannot install on a Resin device

cSchubes, 2018-07-01:
Hi, 

When attempting to deploy this package (npm install) to a Raspberry Pi 3 running resinOS a compilation error occurs:
```
../src/rpi-ws281x.cc: In function ‘void init(const Nan::FunctionCallbackInfo<v8::Value>&)’:
[main]  ../src/rpi-ws281x.cc:100:38: error: no matching function for call to ‘v8::Object::HasOwnProperty(v8::Local<v8::String>&)’
[main]       if(config->HasOwnProperty(symFreq)) {
```
The base image is resin's provided resin/raspberry-pi-node Docker image which has the latest version of node installed. Another issue mentioned it was an issue with node versions, but I cannot change the node version of this image.

Any help is greatly appreciated.

usefulthink, 2018-07-05:
So, I just found out what resinOS even is :/ 
To me the error-message sounds like it is having problems finding either the node header-files or it's shared libraries to link to. Given that it's a containerized thing that would kind-of make sense. Can you check if you can install other native node-modules there?

You could also try to run `node-gyp rebuild --verbose` in the module-folder (`node_modules/rpi-ws281x-native`) to get some more information on the error.

cSchubes, 2018-07-05:
I think it’s actually independent of resin - as mentioned in issue #82, it’s failing to build on a system using node 10. I was able to rollback the version using resin/raspberry-pi-node:8 as the base image and everything works beautifully! I am curious about one thing though: one of the methods provided changes brightness. In what way is this done? Is it just proportional changing of the RGB color channels or is it gamma corrected in any way?

usefulthink, 2018-07-07:
Brightness is directly handed over to the c-library (see here https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.c#L1139 and search for `scale`). It is used to scale all color-channels independently before doing the gamma-correction.

closed 2018-07-07 by usefulthink