beyondscreen/node-rpi-ws281x-native #87: root user and all network connections to the Raspberry "freeze" after POSTing a file to express server while ws281x-animation is running.

HerrZatacke, 2018-07-09:
Hi there,

First of all I have to say I appreciate all the work that is being put into this module. 

I have a strange issue with this library in combination with [express](https://www.npmjs.com/package/express) (the node http server) where it seems **the root-user kind of "gets locked" or "freezes"** and all **network connections seem to die** until the next restart **after POSTing a file** to an express-server running on the Raspberry.

_I'll try to give a very detailed description even if some of the following information might be irrelevant._
_If there's anything I can do to dig further into this issue, let me know (I think i have high knowledge of the node.js universe, but [bindings](https://npmjs.com/package/bindings) totally scare me ;-)_

### My Setup
* Raspberry 3 B+
* running raspbian installed via NOOBS
* node `v8.11.3` installed like this https://www.w3schools.com/nodejs/nodejs_raspberrypi.asp
* npm version `6.1.0`
* connected an [8-LED Module](https://www.aliexpress.com/item//32340567635.html) to GPIO 18 like in [this tutorial](https://tutorials-raspberrypi.de/raspberry-pi-ws2812-ws2811b-rgb-led-streifen-steuern/)

### The error
* All network connections from/to the Raspberry are dead.
* All `sudo` commands (used via a local console) do not work.
* When pinging I get this (the error message starts ~30s after starting the ping). `ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar` (see more console output below)
* The tty remains unusable even after Ctrl+C
* `ps -A` returns: (run as `pi` as `sudo ps -A` does not work - see the console output below)
* I don't get any direct error message (still might be possible to find something in logfiles I don't know)
* After all this the actual process running the animation (with root privileges) seems still to be running - yet there is no more response on the TTY.
* Also the desktop of the Raspberry is still usable for the `pi`-user (Mouse moves, Keyboard works - that's how i could `ping` and `ps -A`) except for "reboot" or similar "root"-tasks (which forces me to kill the Raspberry by cutting the power)

<details>
 <summary>Detailed output for "ping 8.8.8.8"</summary>

```
pi@raspberrypi:~ $ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar
^C
--- 8.8.8.8 ping statistics ---
22 packets transmitted, 0 received, 100% packet loss, time 40901ms
```
</details>

<details>
 <summary>Detailed output for "ps -A"</summary>

```
  PID TTY          TIME CMD
    1 ?        00:00:02 systemd
    2 ?        00:00:00 kthreadd
    3 ?        00:00:00 kworker/0:0
    4 ?        00:00:00 kworker/0:0H
    5 ?        00:00:00 kworker/u8:0
    6 ?        00:00:00 mm_percpu_wq
    7 ?        00:00:00 ksoftirqd/0
    8 ?        00:00:00 rcu_sched
    9 ?        00:00:00 rcu_bh
   10 ?        00:00:00 migration/0
   11 ?        00:00:00 cpuhp/0
   12 ?        00:00:00 cpuhp/1
   13 ?        00:00:00 migration/1
   14 ?        00:00:00 ksoftirqd/1
   15 ?        00:00:00 kworker/1:0
   16 ?        00:00:00 kworker/1:0H
   17 ?        00:00:00 cpuhp/2
   18 ?        00:00:00 migration/2
[... 121 lines cut ...]
```
</details>

### How to reproduce the error
* download [this gist](https://gist.github.com/HerrZatacke/7c75b03a92776454364b23d022814a13) (there are two separate scripts I combined into one example gist for easier testing - the error also happens when running as two completely different packages in different folders)
* unzip and run `npm i`
* `sudo npm run ws281x` starts the animation
* `npm run express` starts the very simple webserver
* navigate to `http://[IP]:3000`
* select a file (image ~2MB works for me) and click the `!`-button (this starts the upload)
* there should be no more response from the server - the raspberry should be "dead"

### What else I have tried:
* I used the new `v1.x`-branch and incorporated the changes [noted here](https://github.com/beyondscreen/node-rpi-ws281x-native/issues/76) - Same error ocurred
* I used the python examples [from here](https://github.com/jgarff/rpi_ws281x) in combination with the `npm run express` - No Freezes here.
* Running the test animation script without sudo `npm run ws281x` - No Freezes (and no animation, duh ;))
* killing all 'node' processes - Could not kill the node process started by root.
* Sending the file while the animation loop is not running (even while the "root" process is running). - No freezes
* My current workaround is to query if the animation is running before sending a file, but validation purely on the client side is not a good solution I think.

### What I have **not** tried:
* running a different type of webserver (Apache, nginx, etc)
* running a different node version

If you think any of this might help, I'll be happy to try these.

usefulthink, 2018-07-10:
Hey there, you absolutely deserve a 🥇🏆for the most detailed bug reported here yet!

The error-message (`ping: sendmsg: Kein Hauptspeicher für den Puffer verfügbar`, original text `No buffer space available`) sounds suspiciously like the pi ran out of some sort of memory. Further searching leads to this SO-question: https://askubuntu.com/questions/210451/what-does-ping-sendmsg-no-buffer-space-available-mean - there are some things to try in the answers, like resetting the network-interface

Some more things you can try:

 - you can use `sudo su -` to get a full shell with the root-user logged in, that might help with issuing root-commands after the error occured.
 - if `kill`/`killall` alone doesn't help, try `kill -9 $PID`/`killall -9 $NAME`

Would you mind joining me in our slack (https://livejs-slackin.herokuapp.com/) so we can chat about this? I'm happy to help and help to debug this issue, but I'm a bit out of Ideas right now. Having a real-time conversation would probably help..

HerrZatacke, 2018-07-10:
Hi,

Sure! I'll try both of your suggestions and will also join your channel to report back :-)

xoblite, 2018-09-30:
Hi,

_First of all I have to say I appreciate all the work that is being put into this module._

I can only join the choir here - much appreciated indeed! As for the issue, I have been experiencing similar symptoms to the ones described by the OP, but haven't been able to track down the issue yet either (btw, updating to Node.js 8.12 doesn't seem to improve things). May I ask though, how fast is your animation running? (i.e. interval in msecs between frames, or frames per second). @usefulthink, any recommendations here? (nb. this said, I tried decreasing one of my animations from 20 to 100 msec between updates, but that didn't seem to help; the Zero becomes unresponsive but the animation continues).

Work-in-progress in case you want to check it out yourselves: [https://github.com/xoblite/HAP-NodeJS-accessories](https://github.com/xoblite/HAP-NodeJS-accessories) (see Unicorn_pHAT_accessory.js)

Thanks again,

BR//Karl-Henrik

xoblite, 2018-10-05:
After some more testing (including running the same code via Pimoroni's python library and managing heavy load at the same time) I can pretty much confirm the issue is somewhere in the library; possibly in the default (i.e. unspecified / passed on) init settings. What's interesting is that a few, low pps of network traffic is OK, but more than that and it freezes. Could it be a DMA conflict again perhaps? For reference, Pimoroni uses the following init options in their Python library for Unicorn HAT/pHAT:

`# LED strip configuration:
LED_COUNT      = 64      # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 10      # DMA channel to use for generating signal
LED_BRIGHTNESS = 128     # Set to 0 for darkest and 255 for brightest
LED_CHANNEL    = 0       # PWM channel
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_GAMMA = [`

Will dig around a bit more and see if I can pin down the issue, though the (old branched upstream) selection of DMA channel looks like a candidate at this point (cf. other reports on DMA vs RPi3B+?).

Any thoughts?

BR//Karl-Henrik

xoblite, 2018-10-05:
#90

xoblite, 2018-10-05:
#90

xoblite, 2018-10-05:
Update: Tried changing the DMA channel in [rpi-ws281x.cc](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/src/rpi-ws281x.cc) to **10** and recompiling, and everything seems to be working like a charm now (running Node.js, Prometheus, Redis, sudo apt-get update and frequently hitting my own HTTP server code at the same time, seemingly no issues so far at least).

Profe66er, 2018-10-06:
> default DMA channel

Hello, I am having the exact same problem. Could you please give a detailed description on how to change the DMA channel and all the steps needed to get this working?

xoblite, 2018-10-06:
@Profe66er:

> Hello, I am having the exact same problem. Could you please give a detailed description on how to change the DMA channel, how to recompile and all the steps needed to get this working?

I'll try... (I had never done it before either, but I think this is the correct workflow)

- Install [node-gyp](https://github.com/nodejs/node-gyp) (nb. the instructions on that page match my steps below btw).
- Change (e.g. using nano) row 19 of `rpi-ws281x.cc` in the `<ParentFolder>/node_modules/rpi-ws281x-native/src` folder to read `#define DEFAULT_DMANUM          10`, and save.
- Change directory back to `<ParentFolder>/node_modules/rpi-ws281x-native` and perform a `node-gyp rebuild` (this performs a combined node-gyp "clean", "configure" and "build" operation, just in case). After it's finished, cd into `<ParentFolder>/node_modules/rpi-ws281x-native/lib/binding` and check the date on `rpi_ws281x.node`. If all went well, it should now have the date/time the rebuild was finished.
- Restart your related node.js processes (rebooting never hurts after serious issues either though IMO).
- Push the system limits that previously got you into problems, and let us know if this solves the issue for you as well. (Are you on RPi 3B+ or 0W by the way? Raspbian Stretch?)

Clear as mud? ;) ...let me know otherwise.

BR//Karl-Henrik

Profe66er, 2018-10-06:
Thank a lot. Very helpful. Im on RPI 3B+ and I was trying to run a Spotify api program while running the neopixle server but the neopixle server kept on crashing. I was using a Ws2812B too. It all seems to work now but Ill do some more testing. 
Thanks alot!!!

xoblite, 2018-10-10:
@HerrZatacke @Profe66er Any updates?

izzy26, 2018-10-29:
Duh I wish I found this issue last week when I ran into the same problem. My scenario was that I was using my Pi as a WiFi AP (using the built in WiFi chip on wlan0), connecting my phone to it to control the LED strip. It froze every time I tried to serve a static HTML file from the Express. However, I solved the issue by reversing the connection logic - my phone was set to an AP mode and Pi was connected to it (although I have to point out that it was connected using a USB WiFi dongle on wlan1) - then everything worked just fine.
This would point to the direction that the DMA channel 5 is somehow used by the internal WiFi chip too.
Anyhow, thank you very much for providing this solution, might come in handy later! 👍

xoblite, 2019-01-02:
After putting the same type of pHAT on my [Raspberry Pi 3A+](https://www.raspberrypi.org/products/raspberry-pi-3-model-a-plus/) I can confirm that the DMA 5 issue applies to this model as well, and that it's solved by changing the DMA channel to 10 as per above.

FYI, another way (read: perhaps useful beyond waiting for the DMA change specifically to happen upstream) to address this, that I now use myself in my code, is to pass the modified input parameters to the driver as part of the init call:

    // LED DRIVER INIT PARAMETERS - DON'T CHANGE THESE UNLESS YOU HAVE PROBLEMS AND KNOW WHAT YOU'RE DOING...
    const DRIVER_FREQUENCY = 800000;
    const DRIVER_DMA_NUMBER = 10; // Note: The sometimes used DMA 5 causes networking issues on (at least) RPi Zero W, 3B+ and 3A+!
    const DRIVER_GPIO_PIN = 18;
    const DRIVER_INVERT = 0;
    const DRIVER_BRIGHTNESS = 255;

    // ...

    var driverInitObject = new Object(),
        frequency = DRIVER_FREQUENCY,
        dmaNum = DRIVER_DMA_NUMBER,
        gpioPin = DRIVER_GPIO_PIN,
        invert = DRIVER_INVERT,
        brightness = DRIVER_BRIGHTNESS;
    driver.init(numLeds, driverInitObject);

@usefulthink Any update on the subject? Will you pull this one yourself, or do you want me to do it?

NoahRoseLedesma, 2019-01-29:
Im experiencing similar 'freezing' issues. Im using my RPI as a network bridge. Running an animation for an extended period of time kills hostapd and seems to freeze up the system.

nicklasfrahm, 2019-05-30:
I think we should at least update the `README.md` to document this. I suspected my SD cards until I found this issue.

xoblite, 2019-05-30:
@usefulthink Any update on the subject? Will you pull this one yourself, or do you want me to do it? (nb. the change to DMA channel 10 was committed to upstream by jgarff a very long time ago)