beyondscreen/node-rpi-ws281x-native #91: Usage with Node-RED

alberto2000, 2018-08-16:
I'm using your library as a dependency for a custom Node-RED node I'm working on. When I start node-red with no root privilege the program runs but obviously no LEDs light up as the library needs to have root privileges granted for GPIO access - but when I run node-red with sudo I get this error in console: 

> TypeError: setChannelParam(): expected argument 3 to be the value

Any hint on where this stems from?

usefulthink, 2018-08-16:
so you're using the 1.0 version? 

This can happen when one of the option-values has an invalid value (i.e. it's not a number or a boolean). Here is the code that throws that error: https://github.com/beyondscreen/node-rpi-ws281x-native/blob/724303a4518314086fa5765e5c402218c30e8591/src/rpi-ws281x.cc#L105-L109

And this is where it is called:

https://github.com/beyondscreen/node-rpi-ws281x-native/blob/724303a4518314086fa5765e5c402218c30e8591/lib/ws281x-native.js#L105-L110

usefulthink, 2018-08-16:
If you don't, the initialization of the ws281x-library won't happen.

alberto2000, 2018-08-16:
I know - any idea why this line triggers the error only when run as root? Or why it throws at all?

alberto2000, 2018-08-16:
My fault - I was passing 60 as a string (not reflected in the snippet I put above) :|

closed 2018-08-16 by alberto2000