beyondscreen/node-rpi-ws281x-native #94: No LEDs turn on

davidpox, 2019-01-17:
Hi, I'm attempting to create a web interface for controlling WS2812 LEDs. I managed to get the `rainbow.js` example working from the examples folder using `sudo node rainbow.js` and that worked fine. However I am now attempting to port this code over to my own use-case but it isn't working. I copied the `ws281x-native` stuff over and its related dependencies. The library initializes fine (logging `ws281x` in `init()` shows the bindings and functions. 
However, calling `ws281x.render(this.pixelData);` yields no results, no errors. 

This script is being called from `index.html` running on an apache2 webserver on my RPI 3B+. Connections to the webserver work fine. No errors during load, either. 

```js
const NUM_LEDS = 60;
const ws281x = require('./lib/ws281x-native');
ws281x.init(NUM_LEDS);

export default class Main {
	constructor() {
		this.pixelData = new Uint32Array(NUM_LEDS);
	}

	init() {
		console.log(ws281x);
		for(var i = 0; i < NUM_LEDS; i++) {
			this.pixelData[i] = 0xffcc22;
		}
		ws281x.render(this.pixelData);
	}
}
```

Maybe I am misunderstanding how these work. Or I am possibly running it wrong, because I noticed that I run rainbow.js using `sudo node` but not the apache webserver. How could I use these two together?

adamcoulombe, 2019-02-21:
Is apache running with root priviledges? Ultimately the script needs to be run with root.

adamcoulombe, 2019-02-21:
The way I did it was using a express.js server to handle http requests and running that script using root (sudo). I am not as familiar with how it gets called if youre using apache.