beyondscreen/node-rpi-ws281x-native #98: Fix build errors for Node.js v12

Granjow, 2019-08-14:
There are compilation errors on Node.js v12 like:

```
../src/rpi-ws281x.cc:87:52: error: no matching function for call to 'v8::Value::Int32Value()'
   ledstring.channel[0].count = info[0]->Int32Value();
                                                    ^
In file included from /root/.cache/node-gyp/12.8.0/include/node/node.h:63:0,
                 from ../../nan/nan.h:54,
                 from ../src/rpi-ws281x.cc:1:
/root/.cache/node-gyp/12.8.0/include/node/v8.h:2576:40: note: candidate: v8::Maybe<int> v8::Value::Int32Value(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<int32_t> Int32Value(Local<Context> context) const;
                                        ^~~~~~~~~~
/root/.cache/node-gyp/12.8.0/include/node/v8.h:2576:40: note:   candidate expects 1 argument, 0 provided
../src/rpi-ws281x.cc:91:46: error: no matching function for call to 'v8::Value::ToObject()'
     Local<Object> config = info[1]->ToObject();
```

from functions which have been removed in this version.

An update would be very appreciated!

VTsuprov, 2020-01-01:
Is it correct that there is no way to install the library with the latest version on NodeJS (>12.0)? At least I have tried all different forks and all ways of installing the library. Any time it was a fail with compilation

ckirmse, 2020-08-08:
the latest version, 0.10, works with node 12 and higher. This issue should be resolved.

Granjow, 2020-08-09:
Cool, thank you!

closed 2020-08-09 by Granjow