beyondscreen/node-rpi-ws281x-native #100: Random color corruption of WS2813 using 3B+

JeffJassky, 2019-09-10:
**Board**: Raspberry Pi 3b+
**LEDs**: WS2813
**Pin**: GPIO 12 / PWM0
**dmaNum**: 10
**Number of pixels**: 236

Works as expected 80% of the time.

intermittently, random blocks of pixels (sometimes a single pixel, usually larger blocks of pixels) become a seemingly random color and no longer respond to calling `render()`. The surrounding pixels on the rest of the strip respond properly, but the corrupt blocks stay the same color.

`reset()`, reinitializing with `init(numPixels, { dmaNum: 10 });` and re-rendering with an entirely new array of pixel data does not uncorrupt the blocks. The entire strip has to be power cycled in order to regain proper rendering of the corrupt blocks.

I've tried several raspberry pis and get the same result.

I also have `/etc/modprobe.d/blacklist.conf` configured with
```
blacklist snd_bcm2835
blacklist snd_pcm
blacklist snd_timer
blacklist snd_pcsp
blacklist snd
```

Is there something I'm missing about how the GPIO configuration? Is it possible these channels are still being used by something else at the same time?

Thanks!

usefulthink, 2019-09-10:
hmm, these problems are always a bit hard to debug without knowing a lot more about the exact setup. The fact that re-initializing the library is not resetting the state of the LEDs hints at this being some sort of hardware- or signal-issue. I suspect you don't have access to an oscilloscope?

 - How did you wire up the backup-data line (BIN) of the LEDs?
 - what voltages are you using for data and VCC? How is the circuit between the PI and the LEDs built? 
 - is there a probability for electrical noise in the circuit, especially the data-line (long unshielded wires, nearby electrical fields etc)?
 - anything special about where the corrupted segments start/end?

JeffJassky, 2019-09-10:
@usefulthink thanks for responding!

I'm using a mosfet-based logic shifter to 5.05v. Circuit here:
[image]

BIN is not wired up at all. Would you suggest trying that?

Data line is very short. Emf cross-talk/noise very unlikely.

Nothing special about where the corrupt segments are. Random every time.

I do have access to an oscilloscope. I don't know how to use it well but my buddy probably knows exactly what to do.

Any suggestions before I call my buddy over to oscope it?

usefulthink, 2019-09-13:
I don't fully understand the weird english translation of the datasheet (http://www.normandled.com/upload/201808/WS2815%20LED%20Datasheet.pdf), but I think I ÷would try to wire the BIN together with DIN.

usefulthink, 2019-09-13:
It's very confusing. Some diagrams I see in the image-search show the BIN floating, some have them grounded. According to this, it also doesn't really help explaining what you're seeing:
 
[image]

So, it seems the backup-data line is forwarded from the data-in of the previous LED, so it wouldn't explain multiple LEDs getting stuck