beyondscreen/node-rpi-ws281x-native #101: Root-less usage with /dev/gpiomem

madc, 2019-10-29:
I read through #19 and was wondering, if it would now be possible to use `/dev/gpiomem` instead of `/dev/mem`. I found [this discussion](https://raspberrypi.stackexchange.com/a/47022), where it seems to be possible for python.

usefulthink, 2019-10-29:
that discussion doesn't talk about configuring DMA-transfers for spi/pcm/pwm, which is why I believe the gpiomem-device alone might not be enough. 

Also, this module doesn't use /dev/mem at all, it is the upstream-dependency (https://github.com/jgarff/rpi_ws281x) that is doing that. If anywhere, it would need to be fixed there..

usefulthink, 2019-10-29:
note: it appears that it should be possible to use the SPI-mode without root-privilege: https://github.com/jgarff/rpi_ws281x/pull/257

Not sure though if that commit is already in the latests `@next` release.

closed 2019-10-29 by usefulthink