beyondscreen/node-rpi-ws281x-native #102: node-rpi-ws281x-native linked to an old version of your fork of  rpi_ws281 lib

jimbotel, 2019-11-05:
Hi @usefulthink ,

It looks like node-rpi-ws281x-native is linked to an old version of your fork of  rpi_ws281 lib.
See:   https://github.com/beyondscreen/node-rpi-ws281x-native/tree/master/src
so the changes introduced for RPI4 support are not effective in node-rpi-ws281x-native.
Can you please take a look at that?
Thanks!

Jaime.

usefulthink, 2019-11-07:
this is probably true, haven't updated it in a while. Are you using the rpi-ws281x-native@next version? The older version will very likely not receive any further updates.

jimbotel, 2019-11-07:
Hi @usefulthink . No, I'm using v0.9.0 . I didn't test @next (this is currently out of my control, I'm working with another repository that makes use of this one).
I think @JeffJassky tested @next and he was not able to complete the build.

JeffJassky, 2019-11-10:
@jimbotel it has been quite a white since testing. I do not recall what the issue was at the time. It's possible the issue may have been resolved already and it's possible the issue may have been my environment. It should be tested again.

ckirmse, 2020-08-08:
The latest version, 0.10, no longer uses a fork of rpi-ws281x lib, so it should work fine now. This issue should be resolved.

jimbotel, 2020-08-09:
hi @ckirmse . Thanks for letting me know, I didn't know that. I'll give it a try