beyondscreen/node-rpi-ws281x-native #104: Build errors on RaspberryPi4

ChrisVeigl, 2020-05-01:
Hi!
I get lots of build errors when trying to  `npm install rpi-ws281x-native` on a Raspi4 (Raspbian buster 4.19.97). (node-gyp rebuild fails at the nan module.)

I tried different versions of node (10.0.0, 12.13.1, 13.12.0, 14.1.0).
Any ideas how to deal with this?

lubomirmolin, 2020-05-04:
Try my fork, I used to have similar issues. https://github.com/lubomirmolin/rpi-ws281x-nodejs

ChrisVeigl, 2020-05-05:
thank you for your reply!
i tried the npm i - unfortunately with the same errors from node-gyp. 
i attach the errorlog in case you want to take a look!

[errorlog.txt](https://github.com/beyondscreen/node-rpi-ws281x-native/files/4583806/errorlog.txt)

lubomirmolin, 2020-05-08:
Are you trying to install it on Windows machine? I know I had some issues with the platform

ChrisVeigl, 2020-05-09:
no no - it's clear that this couldn't work on a Windows machine ...
I tried to install on a RaspberryPi 4 with a recent version of Raspian Buster.

SimonGAndrews, 2020-05-12:
hi,
   Im getting the same error log as above in this situation.  Also for info
     gyp info using node-gyp@6.1.0
     gyp info using node@12.16.2 | linux | arm
     gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
    (tried python v 2 also but got same error)
 all on rasp pi 3B.
 
I note this discussion on at least the first error (not warning).
https://stackoverflow.com/questions/58219928/node-12-v8-converting-deprecated-code-to-the-maybe-version

Thanks Simon

TheMightyPilot, 2020-06-18:
any update for this issue and how to get this module working?

SimonGAndrews, 2020-06-18:
https://github.com/TheMightyPilot ,  I resorted to going back to node version 10.21.0 .  There are a few libraries including this one that are failing beyond V10.

SimonGAndrews, 2020-06-18:
oh and also (and prob more significantly) went for this library instead "rpi-ws281x" of this one.

TheMightyPilot, 2020-06-18:
Thanks for your reply @SimonGAndrews

I've used NVM to switch to use node 10.21.0 and npm installed the rpi-ws281x package, however it seems running the test I get the same type of error as I did with this ws281x-native:
```
internal/modules/cjs/loader.js:1208
  return process.dlopen(module, path.toNamespacedPath(filename));
                 ^

Error: The module '/home/pi/test/node_modules/rpi-ws281x/build/Release/rpi-ws281x.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1208:18)
    at Module.load (internal/modules/cjs/loader.js:1002:32)
    at Function.Module._load (internal/modules/cjs/loader.js:901:14)
    at Module.require (internal/modules/cjs/loader.js:1044:19)
    at require (internal/modules/cjs/helpers.js:77:18)
    at Object.<anonymous> (/home/pi/test/node_modules/rpi-ws281x/index.js:2:13)
    at Module._compile (internal/modules/cjs/loader.js:1158:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1178:10)
    at Module.load (internal/modules/cjs/loader.js:1002:32)
    at Function.Module._load (internal/modules/cjs/loader.js:901:14)
```

SimonGAndrews, 2020-06-18:
to be honest MightyPilot , I got lost with this type of error and went in circles with node versions and reinstalls of node and the the packages I am using.    The table https://nodejs.org/en/download/releases/   is a usefull reference to check node version against compatible NODE_MODULE_VERSION.   It looks from the error you've got that elements of node > 10.72  (or installed packages) are still hanging around.   You can try googling the key bit of the error "compiled against a different Node.js version using NODE_MODULE_VERSION" and there are some stack overflows that may help.  

But in the end, I gave up and  I used a different NPM package to work with the neopixel LEDs.  I used https://www.npmjs.com/package/rpi-ws281x
(with node 10.21) and it is working very well.

ckirmse, 2020-08-08:
The latest version of this, 0.10, should work fine with node 12 and higher. This issue should be resolved now.

nduc, 2020-10-18:
Master branch is still on version 0.9 and it's not building for me with Node 12.  Where can I get 0.10?

gbkwiatt, 2020-12-28:
Currently it still doesn't work on Node 14. 

Well, 0.10 works fine, however any other 1.x version gives an error on build.

DJHoltkamp, 2021-04-13:
I cannot get this to build due to it not being able to find nan.h. I have tried 8, 10, and 12.

usefulthink, 2021-04-13:
@DJHoltkamp Thanks for the report, I think I might have [found a problem](https://github.com/beyondscreen/node-rpi-ws281x-native/commit/f9c245c7ae12f5413586fb1d545f0338d96d02f5) with the latest release that could have caused that. Would you mind testing again with the version 1.0.1 I just released?

DJHoltkamp, 2021-04-14:
@usefulthink - So doing npm install rpi-ws281x-native@1.0.1, I now fail with the error "napi.h: No such file or directory". So it fixed the other problem, but this still remains or created a new issue.

usefulthink, 2021-04-14:
Thanks! So I'm wondering now, why is it even including napi.h? That shouldn't be required.
I should probably reactivate a raspberry here so I can test it myself and don't have to rely people like you to do this 😅 

@DJHoltkamp If you don't mind, can you do that again with version 1.0.2? I removed the requirement for napi.h and rewrote the module-initializer. Will probably gat my raspberry back to running order so I can test it myself in the future... Sorry about that.

DJHoltkamp, 2021-04-15:
@usefulthink  Well, I got to a whole stack of error now, but the first one is:
../src/rpi-ws281x.cc:227:15: error: ‘New’ was not declared in this scope
   Set(target, New<String>("init").ToLocalChecked(),

usefulthink, 2021-04-15:
sorry about that, I hope that I can get a PI back to running order in the next days so I can debug this myself. It's probably just something small, but I don't want to continue like this.

usefulthink, 2021-04-15:
Ok, so I figured that as it's completely broken now anyway I can also give that another round of blind deployment :D 
I think I just forgot to add the namespace-qualifier for Nan::New and Nan::Set.

I published another version if you still like to try it, otherwise you can probably also use this as an alternative: https://www.npmjs.com/package/rpi-ws281x

DJHoltkamp, 2021-04-15:
Ok, thanks! I imagine this will be an easy fix once you can actually try natively compiling it.

DJHoltkamp, 2021-04-26:
Sorry to ping this again, but that alternate package you posted does not support some brightness features I need. Did this issue of not being able to compile ever get fixed? I just tried compiling again and still got a new batch of errors about GetFunction and setBrightness..

jpacora, 2021-06-14:
Any news? I need to run this on a RPI 3B+

chrisgonzales, 2021-07-18:
Was curious if anything in the works for this. Starting a project and have a rpi 3b+ i was testing with fresh rasp pi os lite, and nodejs 14. I'm getting an error also that seems related to the GetFunction in the same area of the most recent commit. Saw this user make a comment about getting something of his working but the NODE_MODULE implementation was different. 

https://github.com/beyondscreen/node-rpi-ws281x-native/issues/113

psy0rz, 2021-08-04:
> Was curious if anything in the works for this. Starting a project and have a rpi 3b+ i was testing with fresh rasp pi os lite, and nodejs 14. I'm getting an error also that seems related to the [...]

same problem here, but with a raspberry Pi Zero and Nodejs 14, on raspbian 10

psy0rz, 2021-08-04:
I think i fixed it, the question is how did it work at all before? This is the diff:
```
diff --git a/src/rpi-ws281x.cc b/src/rpi-ws281x.cc
index f487b02..deaafba 100755
--- a/src/rpi-ws281x.cc
+++ b/src/rpi-ws281x.cc
@@ -226,19 +226,28 @@ void finalize(const Nan::FunctionCallbackInfo<v8::Value> &info) {
 
 NAN_MODULE_INIT(InitAll) {
   Nan::Set(target, Nan::New<String>("init").ToLocalChecked(),
-    GetFunction(Nan::New<FunctionTemplate>(init)).ToLocalChecked());
+    Nan::GetFunction(Nan::New<FunctionTemplate>(init)).ToLocalChecked());
 
-  Nan::Set(target, Nan::New<String>("setBrightness").ToLocalChecked(),
-    GetFunction(Nan::New<FunctionTemplate>(setBrightness)).ToLocalChecked());
+//  Nan::Set(target, Nan::New<String>("setBrightness").ToLocalChecked(),
+//    Nan::GetFunction(Nan::New<FunctionTemplate>(setBrightness)).ToLocalChecked());
 
-  Nan::Set(target, Nan::New<String>("reset").ToLocalChecked(),
-    GetFunction(Nan::New<FunctionTemplate>(reset)).ToLocalChecked());
+  Nan::Set(target, Nan::New<String>("setParam").ToLocalChecked(),
+    Nan::GetFunction(Nan::New<FunctionTemplate>(setParam)).ToLocalChecked());
+
+  Nan::Set(target, Nan::New<String>("setChannelParam").ToLocalChecked(),
+    Nan::GetFunction(Nan::New<FunctionTemplate>(setChannelParam)).ToLocalChecked());
+
+  Nan::Set(target, Nan::New<String>("setChannelData").ToLocalChecked(),
+    Nan::GetFunction(Nan::New<FunctionTemplate>(setChannelData)).ToLocalChecked());
+
+ // Nan::Set(target, Nan::New<String>("reset").ToLocalChecked(),
+//    Nan::GetFunction(Nan::New<FunctionTemplate>(reset)).ToLocalChecked());
 
   Nan::Set(target, Nan::New<String>("render").ToLocalChecked(),
-    GetFunction(Nan::New<FunctionTemplate>(render)).ToLocalChecked());
+    Nan::GetFunction(Nan::New<FunctionTemplate>(render)).ToLocalChecked());
 
   Nan::Set(target, Nan::New<String>("finalize").ToLocalChecked(),
-    GetFunction(Nan::New<FunctionTemplate>(finalize)).ToLocalChecked());
+    Nan::GetFunction(Nan::New<FunctionTemplate>(finalize)).ToLocalChecked());
 }
 
 NODE_MODULE(addon, InitAll)
```

So GetFunction needed the Nan:: namespace, makes sense.

But some bindings where pointing to non-existing functions (reset and setbrightness), while others where missing. (set*)

Now it compiles and functions in my setup. (raspbian 10 on pi zero with nodejs 14)

rip3rs, 2021-08-16:
any updates on this PR? I to am having some issue on build.

```shell
239 error make: Leaving directory '/home/pi/ext_nisa/node_modules/rpi-ws281x-native/build'
240 error gyp info it worked if it ends with ok
240 error gyp info using node-gyp@7.1.2
240 error gyp info using node@14.17.0 | linux | arm
240 error gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
240 error gyp info spawn /usr/bin/python3
240 error gyp info spawn args [
240 error gyp info spawn args   '/home/pi/ext_nisa/node_modules/node-gyp/gyp/gyp_main.py',
240 error gyp info spawn args   'binding.gyp',
240 error gyp info spawn args   '-f',
240 error gyp info spawn args   'make',
240 error gyp info spawn args   '-I',
240 error gyp info spawn args   '/home/pi/ext_nisa/node_modules/rpi-ws281x-native/build/config.gypi',
240 error gyp info spawn args   '-I',
240 error gyp info spawn args   '/home/pi/ext_nisa/node_modules/node-gyp/addon.gypi',
240 error gyp info spawn args   '-I',
240 error gyp info spawn args   '/home/pi/.cache/node-gyp/14.17.0/include/node/common.gypi',
240 error gyp info spawn args   '-Dlibrary=shared_library',
240 error gyp info spawn args   '-Dvisibility=default',
240 error gyp info spawn args   '-Dnode_root_dir=/home/pi/.cache/node-gyp/14.17.0',
240 error gyp info spawn args   '-Dnode_gyp_dir=/home/pi/ext_nisa/node_modules/node-gyp',
240 error gyp info spawn args   '-Dnode_lib_file=/home/pi/.cache/node-gyp/14.17.0/<(target_arch)/node.lib',
240 error gyp info spawn args   '-Dmodule_root_dir=/home/pi/ext_nisa/node_modules/rpi-ws281x-native',
240 error gyp info spawn args   '-Dnode_engine=v8',
240 error gyp info spawn args   '--depth=.',
240 error gyp info spawn args   '--no-parallel',
240 error gyp info spawn args   '--generator-output',
240 error gyp info spawn args   'build',
240 error gyp info spawn args   '-Goutput_dir=.'
240 error gyp info spawn args ]
240 error gyp info spawn make
240 error gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
240 error ../src/rpi-ws281x.cc: In function ‘void InitAll(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
240 error ../src/rpi-ws281x.cc:229:5: error: ‘GetFunction’ was not declared in this scope
240 error      GetFunction(Nan::New<FunctionTemplate>(init)).ToLocalChecked());
240 error      ^~~~~~~~~~~
240 error ../src/rpi-ws281x.cc:229:5: note: suggested alternative:
240 error In file included from ../../nan/nan.h:286,
240 error                  from ../src/rpi-ws281x.cc:1:
240 error ../../nan/nan_maybe_43_inl.h:66:33: note:   ‘Nan::GetFunction’
240 error  inline MaybeLocal<v8::Function> GetFunction(
240 error                                  ^~~~~~~~~~~
240 error ../src/rpi-ws281x.cc:232:44: error: ‘setBrightness’ was not declared in this scope
240 error      GetFunction(Nan::New<FunctionTemplate>(setBrightness)).ToLocalChecked());
240 error                                             ^~~~~~~~~~~~~
240 error ../src/rpi-ws281x.cc:235:44: error: ‘reset’ was not declared in this scope
240 error      GetFunction(Nan::New<FunctionTemplate>(reset)).ToLocalChecked());
240 error                                             ^~~~~
240 error ../src/rpi-ws281x.cc:235:44: note: suggested alternative: ‘tzset’
240 error      GetFunction(Nan::New<FunctionTemplate>(reset)).ToLocalChecked());
240 error                                             ^~~~~
240 error                                             tzset
240 error In file included from ../../nan/nan.h:58,
240 error                  from ../src/rpi-ws281x.cc:1:
240 error ../src/rpi-ws281x.cc: At global scope:
240 error /home/pi/.cache/node-gyp/14.17.0/include/node/node.h:758:43: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
240 error        (node::addon_register_func) (regfunc),                          \
240 error                                            ^
240 error /home/pi/.cache/node-gyp/14.17.0/include/node/node.h:792:3: note: in expansion of macro ‘NODE_MODULE_X’
240 error    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
240 error    ^~~~~~~~~~~~~
240 error ../src/rpi-ws281x.cc:244:1: note: in expansion of macro ‘NODE_MODULE’
240 error  NODE_MODULE(addon, InitAll)
240 error  ^~~~~~~~~~~
240 error make: *** [rpi_ws281x.target.mk:114: Release/obj.target/rpi_ws281x/src/rpi-ws281x.o] Error 1
240 error gyp ERR! build error
240 error gyp ERR! stack Error: `make` failed with exit code: 2
240 error gyp ERR! stack     at ChildProcess.onExit (/home/pi/ext_nisa/node_modules/node-gyp/lib/build.js:194:23)
240 error gyp ERR! stack     at ChildProcess.emit (events.js:376:20)
240 error gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
240 error gyp ERR! System Linux 5.10.17-v7l+
240 error gyp ERR! command "/usr/bin/node" "/home/pi/ext_nisa/node_modules/.bin/node-gyp" "rebuild" "-j4" "--release"
240 error gyp ERR! cwd /home/pi/ext_nisa/node_modules/rpi-ws281x-native
240 error gyp ERR! node -v v14.17.0
240 error gyp ERR! node-gyp -v v7.1.2
240 error gyp ERR! not ok
```

DJHoltkamp, 2021-08-22:
Additionally would love to see a PR which fixes this issue. Seems to make the project unusable.

FeBe95, 2021-08-22:
I got it working by changing the file `rpi-ws281x.cc` according to the comment by @psy0rz (see https://github.com/beyondscreen/node-rpi-ws281x-native/issues/104#issuecomment-893007649):

> I think i fixed it, the question is how did it work at all before? This is the diff:
> ...

Unfortunatly the `setBrightness()` function was essential for my use case and sadly isn't available anymore. What's the reason for this change? Did the library [rpi_ws281x](https://github.com/jgarff/rpi_ws281x) by Jeremy Garff remove this feature?

DJHoltkamp, 2021-08-22:
Wow, this is very disappointing to hear. I have another library working, but I specifically wanted to get this library working because I could find no other replacement that allowed me to set the brightness of the LEDs. If you or anyone else knows how to fix this or an alternate library, please let me know.

gbkwiatt, 2021-08-22:
I've made it work with node 14, on 1.x branch and If I remember correctly I used set Brightness, However you can write your own method for setBrightness, pretty much just add multiplier  for R G B and check for max 255 and lowest. 0 https://github.com/gbkwiatt/node-rpi-ws281x-native

This is what I've done with it
https://www.youtube.com/watch?v=kDIB8_dzaEI

FeBe95, 2021-08-22:
Thanks for the code examples. I will probably end up writing my own `setBrightness` method now as well. I need to have it separate from the normal `render` method as it would conflict with it otherwise. I hooked up a lux sensor in order to dim all LEDs simultaneously but independently from the selected colors. It might need some tweaking and hacking here and there now.
A global `setBrightness` method was perfect for my setup :/

DJHoltkamp, 2021-08-22:
This is somewhat helpful, but perhaps this library or this protocol simply does not support what I thought it did. I thought that this supported using brightness within the color array itself as a 4th channel to say how much power actually goes to the LED to produce that color. For instance, if I wanted it to be fully red, but very dim, I could specify this.

You would then push this through the library like this:

`        let color = (brightness << 24) | (red << 16) | (green << 8)| blue;
        pixels[i + offset_light] = color;`

Does something like this not exist, or does the protocol not work this way?

FeBe95, 2021-08-22:
Okay, I think a found a way to change the global brightness!

```javascript
function setBrightness(channel, value) {
    channel.brightness = value
    ws281x.render();
}
// ...
var channel = ws281x(NUM_LEDS, options);

setBrightness(channel, 50);
```

This seems to work just fine with the new 1.x version of this project.

DJHoltkamp, 2021-08-22:
Ok, this entire time, I have been chasing the wrong issue. I assumed the lack of brightness settings was not letting me get smooth linear color interpolation when trying to make a smooth animation between lights on my strip. This is why I thought your brightness solution wasn't working. When I tried to interpolate between the LED's to animate, it would have a bit of flashing where it would get brighter and darker as it transitioned. 

After all of this searching, I finally figured out the issue:  **There is no gamma correction in this library**

Once I figured it out, it looks 100% perfect. I'm just frustrated it took me this long. For the record, this is how you would do it:

`        
function gammaAdjust(value, gamma)
{
    let gammaCorrection = 1 / gamma;
    let color = Math.pow((value / 255), gammaCorrection) * 255;
    return color;
}

red = gammaAdjust(red, 2.2);
green = gammaAdjust(green, 2.2);
blue = gammaAdjust(blue, 2.2);

let color = (brightness << 24) | (red << 16) | (green << 8) | blue;
pixels[i + offset_light] = color;
`

Thanks for all the help getting this building again though!

gbkwiatt, 2021-08-23:
Ah fair maybe I should use gamma correction. I do use color lerp / interpolation and it works pretty well.

Deffinitely library could be maintained, it's pretty cool project and since i proficient in JS and pretty much everything can be ported from C to node, sky is the limit and I love it.

psy0rz, 2021-08-24:
Please keep an eye on https://github.com/psy0rz/rpi-ws281x-smi

It allows you to control 16 ledstrips in parallel on all raspberry pi types.