beyondscreen/node-rpi-ws281x-native #107: Raspberry Pi 4 bcm2711

westerxddd, 2020-08-20:
I had to add: **case 'bcm2711': return 1;** 
to: **lib/ws281x-native.js**

```
var raspberryVersion = (function() {
        var cpuInfo = require('fs').readFileSync('/proc/cpuinfo').toString(),
            socFamily = cpuInfo.match(/hardware\s*:\s*(bcm\d+)/i);
            console.log(cpuInfo.match(/hardware\s*:\s*(bcm\d+)/i));
        if(!socFamily) { return 0; }

        switch(socFamily[1].toLowerCase()) {
            case 'bcm2708': return 1;
            case 'bcm2835': return 1;
            case 'bcm2709': return 2;
            case 'bcm2711': return 1; <--- this line to work with my raspberry pi 4
            default: return 0;
        }
    } ());
```

CaskAle, 2020-08-20:
I discovered the same and found that the same case statement with a return of 2 rather than 1 works as well.  I was not able to determine what the return value is used for??

westerxddd, 2020-08-20:
@CaskAle me too.

So i suspect this is only to limit script only for raspberry use.

dJPoida, 2020-09-21:
I also had to add this line on a Raspberry Pi 4. Does the return value perhaps describe the device GPIO pinout?

Deboracgs, 2020-11-01:
> I had to add: **case 'bcm2711': return 1;**
> to: **lib/ws281x-native.js**
>
> ```
> var raspberryVersion = (function() {
>         var cpuInfo = [...]

it's really works! thank you!