beyondscreen/node-rpi-ws281x-native #113: Initialize with different options

gbkwiatt, 2020-12-28:
I am having difficulties trying to make it work. 
I used C library previously with settings for my stripe:
```C
#define LED_PIN 19 //#GPIO pin connected to the pixels(18 uses PWM !).
//LED_PIN         10      // # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
#define LED_FREQ_HZ 800000 //#LED signal frequency in hertz(usually 800khz)
#define LED_DMA 10         // # DMA channel to use for generating signal (try 10)
#define LED_BRIGHTNESS 255 // # Set to 0 for darkest and 255 for brightest
#define LED_INVERT 0       //True to invert the signal(when using NPN transistor level shift)
#define LED_CHANNEL 1      // # set to '1' for GPIOs 13, 19, 41, 45 or 53
```
That was working fine in Cpp

No I am trying to convert that to this library but I can't make it to work.
I can't test it on default gpio 18 because I broke it ;)

how do you properly set it up ?  Tried that:
```javascript
const options = {
        dma: config.leds.dma,
        freq: config.leds.freq,
        channels: [
            { count: 20, gpio: 19, invert: false, brightness: 255, stripType: 'ws2812' },
            { count: 20, gpio: 19, invert: false, brightness: 128, stripType: 'ws2812' }
        ]
    };
```

and that:
```javascript
const options = {
        dma: config.leds.dma,
        freq: config.leds.freq,
        gpio: 19,//config.leds.gpio,
        invert: config.leds.invert,
        brightness: config.leds.brightness,
        stripType: 'ws2812'
    };
```

gbkwiatt, 2020-12-28:
OK I've made v1.x  work with node14. 

Had to update `"nan": "^2.14.2",` to the latest in package.json 

And because some functions are deprecated I had to modify `rpi-ws281x.cc` as below.
Then `npm install` works, and I was amble to initilize LEDs like this (and use channel 1 for GPIO 19): 

```javascript
 const options = {
        dma: config.leds.dma,
        freq: config.leds.freq,
        channels: [
            { count: 20, gpio: 18, invert: false, brightness: 255, stripType: ws281x.stripType.WS2812 },
            { count: 20, gpio: 19, invert: false, brightness: 255, stripType: ws281x.stripType.WS2812 }
        ],
    };
    // ---- trap the SIGINT and reset before exit
    process.on('SIGINT', function () {
        ws281x.reset();
        log.debug("Reseting Leds on exit...")
        process.nextTick(function () { process.exit(0); });
    });
    const channel = ws281x.init(options);
// Rainbow test on channel 1
    var offset = 0;
    setInterval(function () {
        for (var i = 0; i < 20; i++) {
          channel[1].array[i] = colorwheel((offset + i) % 256);
        }
        offset = (offset + 1) % 256;

        ws281x.render();
      }, 1000 / 30);
```

```cpp
//rpi-ws281x.cc
#include <nan.h>

#include <v8.h>

#include <stdio.h>
#include <stdint.h>
#include <string.h>

#include <algorithm>

extern "C" {
#include "rpi_ws281x/ws2811.h"
}

using namespace v8;

#define DEFAULT_TARGET_FREQ 800000
#define DEFAULT_GPIO_PIN 18
#define DEFAULT_DMANUM 10

#define PARAM_FREQ 1
#define PARAM_DMANUM 2
#define PARAM_GPIONUM 3
#define PARAM_COUNT 4
#define PARAM_INVERT 5
#define PARAM_BRIGHTNESS 6
#define PARAM_STRIP_TYPE 7

ws2811_t ws281x;

/**
 * ws281x.setParam(param:Number, value:Number)
 * wrap setting global params in ws2811_t
[... 213 lines cut ...]

  ws2811_fini(&ws281x);
}

/**
 * initializes the module.
 */
void initialize(Local<Object> exports)
{
  ws281x.freq = DEFAULT_TARGET_FREQ;
  ws281x.dmanum = DEFAULT_DMANUM;

  NAN_EXPORT(exports, setParam);
  NAN_EXPORT(exports, setChannelParam);
  NAN_EXPORT(exports, setChannelData);
  NAN_EXPORT(exports, init);
  NAN_EXPORT(exports, render);
  NAN_EXPORT(exports, finalize);
}

NODE_MODULE(rpi_ws281x, initialize)

// vi: ts=2 sw=2 expandtab

```

ngrealish, 2022-10-04:
I am trying 19 on channel 1 with no results unfortunately.  I've used your update to the rpi-ws281x.cc with no success.  Any other hints or thoughts on this one?  I cannot use 13.  18 is working for me, so I'd like to use 19 as a second set.