beyondscreen/node-rpi-ws281x-native #120: [v1.x] config issue;

Toshimonster, 2021-03-21:
configs seem volatile, and bleeds onto led channel 1. As such,
`options = {
        dma: 10,
        freq: 800000,
        channels: [
            { count: 10, gpio: 18, invert: false, brightness: 255, stripType: w.stripType.WS2812 },
            { count: 256, gpio: 13, invert: false, brightness: 10, stripType: w.stripType.WS2812 }
        ],
    };`
will result in channel 1 giving nothing but the first 10 pixels, and 
`options = {
        dma: 10,
        freq: 800000,
        channels: [
            { count: 256, gpio: 18, invert: false, brightness: 255, stripType: w.stripType.WS2812 },
            { count: 256, gpio: 13, invert: false, brightness: 10, stripType: w.stripType.WS2812 }
        ],
    };`
works fine. In addition, it seems to be the lowest of the two.
In addition, this seems to happen with brightness too:
`options = {
        dma: 10,
        freq: 800000,
        channels: [
            { count: 256, gpio: 18, invert: false, brightness: 0, stripType: w.stripType.WS2812 },
            { count: 256, gpio: 13, invert: false, brightness: 10, stripType: w.stripType.WS2812 }
        ],
    };`
will show nothing.
I've had a look, and on the javascript side of things, the parameters are maintained. This seems to be an issue with the binding.

usefulthink, 2021-03-22:
Thanks for the report, I never actually worked with multiple LED strips, so that is quite valuable. 
But I think this is something in the upstream-library. This part of the code shouldn't be causing this:

https://github.com/beyondscreen/node-rpi-ws281x-native/blob/487204d64f273801a69b1eb45d2a93f2ffcbad39/src/rpi-ws281x.cc#L111-L131

Can you maybe try to do the same with the upstream-library (https://github.com/jgarff/rpi_ws281x) and see if that behaves any diffferent? I can't see anything obvious there either, but maybe there is a better place to solve this.

usefulthink, 2021-03-23:
I think I found the issue, more or less by accident: https://github.com/beyondscreen/node-rpi-ws281x-native/blob/487204d64f273801a69b1eb45d2a93f2ffcbad39/src/rpi-ws281x.cc#L182-L184

I published a fix with the latest version, `1.0.0-alpha3` (49c7018da34df5cd8f9d653ba0f17f8141365327).

closed 2021-03-23 by usefulthink