beyondscreen/node-rpi-ws281x-native #121: RPi3 B+ Rev 1.3 throws 'hardware revision not supported' error. Everything works on RPi3 B Rev 1.2

aqmattil, 2021-03-23:
I’m running rpi-ws281-native@1.0.0-alpha1 (I need to control two led strips with one device) on two 'almost' identical Raspberries, with the only difference being the revision number. On the older version everything works, but with the newer one I get the ‘hardware revision not supported’ error.

**Everything works**

```
cat /proc/cpuinfo
…

Hardware        : BCM2835
Revision        : a02082
Model           : Raspberry Pi 3 Model B Rev 1.2
```

Node 8.17, Raspbian Jessie

**Hardware revision not supported**

```
cat /proc/cpuinfo
…
Hardware        : BCM2835
Revision        : a020d3
Model           : Raspberry Pi 3 Model B Plus Rev 1.3

```
Node 8.17, Raspbian Buster (tried also with Jessie and Stretch to see if this was a compiler issue)

I’ve tried git cloning https://github.com/jgarff/rpi_ws281x.git to ./node_modules/rpi-ws281x-native/src/ as per (the old thread) [https://github.com/beyondscreen/node-rpi-ws281x-native/issues/76](https://github.com/beyondscreen/node-rpi-ws281x-native/issues/76), and running npm install but this didn’t help. 

I've also tried uninstalling/reinstalling the rpi_ws281x python library (pip3 uninstall rpi_ws281x / pip3 install rpi_ws281x) and running npm install both in the rpi-ws281x-native folder as well as the entire project.

All hints would be greatly appreciated, as currently I'm banging my head against the wall quite badly.

aqmattil, 2021-03-23:
I managed to sort this out, so I'm posting this if anyone comes up with the same issue. I'm running this in a Docker container, and managed to sort this out by doing the following in my Dockerfile:

```
# Install npm modules for the application
RUN JOBS=MAX npm install \
 --production --unsafe-perm \
  && npm cache clean --force \
  && rm -rf /tmp/* \
  && node_modules/.bin/electron-rebuild

# try a fix to get rpi-ws281x-native working
RUN cd ./node_modules/rpi-ws281x-native/src \
  && rm -rf rpi_ws281x \
  && git clone https://github.com/jgarff/rpi_ws281x.git \
  && npm install

# and rebuild electron again
RUN JOBS=MAX ./node_modules/.bin/electron-rebuild

```

closed 2021-03-23 by aqmattil

usefulthink, 2021-03-23:
well, the good news is that I just did basically the same thing, including some other smaller changes. It's now up on npm with the latest version of the upstream-library. Install via `npm install rpi-ws281x-native@next` (installed version should be v1.0.0-alpha2).

I would be very glad if you could test this for me really quick as I currently don't have a working setup around...

aqmattil, 2021-03-24:
Hi, thanks for letting me know. I ran a quick test:

package.json
`    "rpi-ws281x-native": "1.0.0-alpha2",
`

and removed the git clone part from my Dockerfile and used straight npm install instead.

I can confirm that this works.

usefulthink, 2021-03-24:
Excellent, thanks a lot. (and great to hear it works) :D

Have fun!

usefulthink, 2021-03-24:
(as a sidenote, I decided it's about time to finally release the 1.0-version, which is now what you'll get when you just install without a release-tag (or with `npm i rpi-ws281x-native@latest`)