beyondscreen/node-rpi-ws281x-native #123: MAX_CHANNELS set to 2?

bweis, 2021-08-05:
Is there a reason that the maximum number of channels is only 2?  I haven't tested it with more yet, figured I should ask before I move forward with testing.

Thanks!

psy0rz, 2021-08-06:
Its inherit to this system, 2 is the max. But we're working on an SMI version that can do 8 or 16 channels parallel.

bweis, 2021-08-06:
I see.  Perhaps you could help me out.  Is there an avenue for controlling 4 independent light strips from 4 different GPIO pins on the raspberry pi?

I would love to not have to string all 4 of them together... 😅

psy0rz, 2021-08-08:
Yes, the only way is via something called SMI, i'm working on it currently: 

https://github.com/psy0rz/rpi-ws281x-smi

westonmiller, 2021-12-15:
> Yes, the only way is via something called SMI, i'm working on it currently:
> 
> https://github.com/psy0rz/rpi-ws281x-smi

I'm trying to add a channel to my Christmas lights setup. Is it possible to just swap out the `node-rpi-ws281x-native` with the smi version?  I'm not seeing much in the readme.

psy0rz, 2021-12-16:
Nope its not a drop in repacement.

sorry you have to look in the sources to see how to use it for now.

westonmiller, 2021-12-19:
Ok, Thank you.  Do you know have to convert an RGB ws2812 signal to a ws2811.  I tried just sending it GRB instead, which mostly works, but the blues look more purple.  (This is related to the MAX CHANNEL limitation as I'm having to add a different type of strand to the end of a ws2812b strand)?

psy0rz, 2021-12-20:
no sorry

closed 2024-07-09 by bweis