beyondscreen/node-rpi-ws281x-native #125: Can't install on Pi Zero with Node 16.6.2.

chrisgwgreen, 2021-09-04:
Getting the following error installing on Raspberry Pi OS Lite for Zero with Node 16.6.2.

> pi@raspberrypi:~/x $ node -v
> v16.6.2

> pi@raspberrypi:~/x $ git clone --recursive https://github.com/beyondscreen/node-rpi-ws281x-native.git
> Cloning into 'node-rpi-ws281x-native'...
...

> pi@raspberrypi:~/x $ cd node-rpi-ws281x-native/
> pi@raspberrypi:~/x/node-rpi-ws281x-native $ npm install

> > rpi-ws281x-native@1.0.2 install
> > node-gyp rebuild -j4 --release
> 
> gyp info it worked if it ends with ok
> gyp info using node-gyp@7.1.2
> gyp info using node@16.6.2 | linux | arm
> gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
> (node:4060) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
> (Use `node --trace-deprecation ...` to show where the warning was created)
> gyp info spawn /usr/bin/python3
> gyp info spawn args [
> gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
> gyp info spawn args   'binding.gyp',
> gyp info spawn args   '-f',
> gyp info spawn args   'make',
> gyp info spawn args   '-I',
> gyp info spawn args   '/home/pi/x/node-rpi-ws281x-native/build/config.gypi',
> gyp info spawn args   '-I',
> gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
> gyp info spawn args   '-I',
> gyp info spawn args   '/home/pi/.cache/node-gyp/16.6.2/include/node/common.gypi',
> gyp info spawn args   '-Dlibrary=shared_library',
> gyp info spawn args   '-Dvisibility=default',
> gyp info spawn args   '-Dnode_root_dir=/home/pi/.cache/node-gyp/16.6.2',
> gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
> gyp info spawn args   '-Dnode_lib_file=/home/pi/.cache/node-gyp/16.6.2/<(target_arch)/node.lib',
> gyp info spawn args   '-Dmodule_root_dir=/home/pi/x/node-rpi-ws281x-native',
> gyp info spawn args   '-Dnode_engine=v8',
> gyp info spawn args   '--depth=.',
> gyp info spawn args   '--no-parallel',
> gyp info spawn args   '--generator-output',
> gyp info spawn args   'build',
> gyp info spawn args   '-Goutput_dir=.'
> gyp info spawn args ]
> gyp info spawn make
> gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
> make: Entering directory '/home/pi/x/node-rpi-ws281x-native/build'
>   ACTION binding_gyp_rpi_ws281x_version_target_make_version_h src/rpi_ws281x/version.h
>   TOUCH Release/obj.target/rpi_ws281x-version.stamp
>   CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/dma.o
>   CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/mailbox.o
>   CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/pcm.o
>   CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/pwm.o
>   CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/rpihw.o
>   CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/ws2811.o
>   AR(target) Release/obj.target/rpi_libws2811.a
>   COPY Release/rpi_libws2811.a
>   CXX(target) Release/obj.target/rpi_ws281x/src/rpi-ws281x.o
> ../src/rpi-ws281x.cc: In function ‘void InitAll(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
> ../src/rpi-ws281x.cc:229:5: error: ‘GetFunction’ was not declared in this scope
>      GetFunction(Nan::New<FunctionTemplate>(init)).ToLocalChecked());
>      ^~~~~~~~~~~
> ../src/rpi-ws281x.cc:229:5: note: suggested alternative:
> In file included from ../node_modules/nan/nan.h:284,
>                  from ../src/rpi-ws281x.cc:1:
> ../node_modules/nan/nan_maybe_43_inl.h:66:33: note:   ‘Nan::GetFunction’
>  inline MaybeLocal<v8::Function> GetFunction(
>                                  ^~~~~~~~~~~
> ../src/rpi-ws281x.cc:232:44: error: ‘setBrightness’ was not declared in this scope
>      GetFunction(Nan::New<FunctionTemplate>(setBrightness)).ToLocalChecked());
>                                             ^~~~~~~~~~~~~
> ../src/rpi-ws281x.cc:235:44: error: ‘reset’ was not declared in this scope
>      GetFunction(Nan::New<FunctionTemplate>(reset)).ToLocalChecked());
>                                             ^~~~~
> ../src/rpi-ws281x.cc:235:44: note: suggested alternative: ‘tzset’
>      GetFunction(Nan::New<FunctionTemplate>(reset)).ToLocalChecked());
>                                             ^~~~~
>                                             tzset
> In file included from ../node_modules/nan/nan.h:56,
>                  from ../src/rpi-ws281x.cc:1:
> ../src/rpi-ws281x.cc: At global scope:
> /home/pi/.cache/node-gyp/16.6.2/include/node/node.h:806:43: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
>        (node::addon_register_func) (regfunc),                          \
>                                            ^
> /home/pi/.cache/node-gyp/16.6.2/include/node/node.h:840:3: note: in expansion of macro ‘NODE_MODULE_X’
>    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
>    ^~~~~~~~~~~~~
> ../src/rpi-ws281x.cc:244:1: note: in expansion of macro ‘NODE_MODULE’
>  NODE_MODULE(addon, InitAll)
>  ^~~~~~~~~~~
> make: *** [rpi_ws281x.target.mk:116: Release/obj.target/rpi_ws281x/src/rpi-ws281x.o] Error 1
> make: Leaving directory '/home/pi/x/node-rpi-ws281x-native/build'
> gyp ERR! build error
> gyp ERR! stack Error: `make` failed with exit code: 2
> gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
> gyp ERR! stack     at ChildProcess.emit (node:events:394:28)
> gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
> gyp ERR! System Linux 5.10.52+
> gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "-j4" "--release"
> gyp ERR! cwd /home/pi/x/node-rpi-ws281x-native
> gyp ERR! node -v v16.6.2
> gyp ERR! node-gyp -v v7.1.2
> gyp ERR! not ok

mukuljha16, 2021-10-06:
Hi @chrisgwgreen,  I'm also having the same issue, did you managed to get it fixed?
Thanks!

anzbert, 2021-10-21:
i have the same issue on a pi4. compilation errors from node-gyp