beyondscreen/node-rpi-ws281x-native #127: Raspberry Pi 4 Model B Rev 1.4 - Hardware not Supported Issue

parrfolio, 2022-02-12:
I'm currently facing this issue in my node app. Running v17.4.0 and NPM 8.3.1 on a Raspberry Pi 4 Model B Rev 1.4. I'm having a difficult time getting past the bindings error. Is this library supported with this version of Raspi?

```
/ws281x-native.js:160
  bindings.init();
           ^

Error: Hardware revision is not supported
```

Note, when I run rpi-ws281x-python library examples my LEDs light right up. No errors. Only seeing errors in the Node JS implementation.

Sure could use some help from anyone who has faced this with this particular model. I've read and tried a lot of solutions but cannot get past this bindings issue.

parrfolio, 2022-02-15:
Fixed this for Pi 4 Model B Rev 14. on Node 17.5 by 

1. Using https://github.com/gbkwiatt/node-rpi-ws281x-native#readme fork
2. Then reinstalling rpi_ws281x library by following these instructions here https://github.com/ibmtjbot/tjbot/issues/158#issue-574903273

This combination of fork plus updating rpi_ws281x worked for me! Sharing for those struggling to use this library as I did. 

More detailed instructions:
`//install from package manager in node_modules
npm install @gbkwiatt/node-rpi-ws281x-native

//install outside project
npm install rpi-ws281x --save 

// cd to
~/node_modules/@gbkwiatt/node-rpi-ws281x-native/src

// delete rpi_ws281x
rm -rf rpi_ws281x/

// copy 
cp -r ~/rpi_ws281x ~/node_modules/@gbkwiatt/node-rpi-ws281x-native/src

// cd out to main folder
cd ~/node_modules/@gbkwiatt/node-rpi-ws281x-native/

// run rebuild
node-gyp rebuild`

harry-finch, 2025-03-10:
I'm trying to implement your solution, but the `node-gyp rebuild` command is not working. It's telling me `gyp: binding.gyp not found`. The file does not exist. Did you get the same error?