beyondscreen/node-rpi-ws281x-native #130: Raspberry Pi 4 Model B Rev 1.5: Hardware revision is not supported

dimzeta, 2022-09-20:
I am running successfully my node app on Raspberry 3 and Raspberry Pi 4 Rev 1.4, but I have this error on Raspberry Pi 4 Rev 1.5:

```
/var/app/node_modules/rpi-ws281x-native/lib/ws281x-native.js:160
  bindings.init();
           ^

Error: Hardware revision is not supported
    at init (/var/app/node_modules/rpi-ws281x-native/lib/ws281x-native.js:160:12)
    at module.exports (/var/app/node_modules/rpi-ws281x-native/lib/ws281x-native.js:206:21)
    at Object.<anonymous> (/var/app/ws281x.js:27:17)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
```

dimzeta, 2022-09-21:
I followed the steps from https://github.com/beyondscreen/node-rpi-ws281x-native/issues/127 and now it is working well.

I didn't have to use @gbkwiatt's fork, so I just did:

```sh
git clone https://github.com/jgarff/rpi_ws281x.git
rm -rf /var/app/node_modules/rpi-ws281x-native/src/rpi_ws281x
cp -r rpi_ws281x /var/app/node_modules/rpi-ws281x-native/src
cd /var/app/node_modules/rpi-ws281x-native
sudo node-gyp rebuild
```

djipco, 2022-12-29:
Is there an ETA for a fix to this issue? The solution from @dimzeta worked for me too so it should be fairly straightforward to implement. 

By the way, thanks for the library!

HarrisZeng0631, 2024-01-12:
> I followed the steps from #127 and now it is working well.
> 
> I didn't have to use @gbkwiatt's fork, so I just did:
> 
> ```shell
> git clone https://github.com/jgarff/rpi_ws281x.git
> rm -rf /var/app/node_modules/rpi-ws281x-native/src/rpi_ws281x
> cp -r rpi_ws281x /var/app/node_modules/rpi-ws281x-native/src
> cd /var/app/node_modules/rpi-ws281x-native
> sudo node-gyp rebuild
> ```

@dimzeta  It works for me. Big helpful. Thanks.

Teslafly, 2024-08-20:
I was unable to get this to work and spent several hours trying things. Unfortunately I am not much into the node ecosystem and only am trying to do this to get https://github.com/bbx10/node-opc-server working. and that's just so I can write to it using a python program that I can't use sudo with. All this just to not need sudo.

harry-finch, 2025-03-10:
I'm trying to implement your solution, but the `node-gyp rebuild` command is not working. It's telling me `gyp: binding.gyp not found`. The file does not exist. Did you get the same error?